bblanchon/ArduinoJson #122: Please add following json encoding

mozgy, 2015-09-19:
Lets say that
JsonObject& data = { "foo":1, "nested": [ { "bar":2 } { "bar":3 } ... ] };
JsonArray& newdata = json.createArray();
JsonArray& tmp = data["nested"];

please add that following works
newdata.add( tmp[0] );

after #120 fix it's possible to do that with
String strTmp; tmp[0]=printTo( strTmp ); newdata.add( strTmp);
but it looks a bit overkill using String buffer copy for that

TIA,

bblanchon, 2015-09-22:
I think it should already work.
Maybe I don't understand.
Can you post a program that demonstrate the issue?

mozgy, 2015-09-23:
For starters it won't even compile ..
call of overloaded 'add(ArduinoJson::JsonArraySubscript)' is ambiguous

bblanchon, 2015-09-29:
Fix in [version 5.0.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.0.4)

closed 2015-09-29 by bblanchon

mozgy, 2015-10-01:
ty