bblanchon/ArduinoJson #123: String.c_str() has no member

creativeknights2012, 2015-09-22:
Using a Intel Edison microcontroller,

I've picked these errors using Arduino 1.6.5:

In file included from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:10:0,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonVariant.hpp:16,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:14,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson.h:7,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:13,
                 from DroneGY80Final.ino:10:
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp: In constructor 'ArduinoJson::JsonObjectKey::JsonObjectKey(const String&)':
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp:17:48: error: 'const class String' has no member named 'c_str'
   JsonObjectKey(const String& key) : _data(key.c_str()) {}
                                                ^
In file included from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9:0,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson.h:7,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:13,
                 from DroneGY80Final.ino:10:
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonArray& ArduinoJson::JsonBuffer::parseArray(const String&, uint8_t)':
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:70:28: error: 'const class String' has no member named 'c_str'
     return parseArray(json.c_str(), nesting);
                            ^
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonObject& ArduinoJson::JsonBuffer::parseObject(const String&, uint8_t)':
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:93:29: error: 'const class String' has no member named 'c_str'
     return parseObject(json.c_str(), nesting);
                             ^
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'char* ArduinoJson::JsonBuffer::strdup(const String&)':
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:100:55: error: 'const class String' has no member named 'c_str'
   char *strdup(const String &src) { return strdup(src.c_str(), src.length()); }
                                                       ^
In file included from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/JsonArray.hpp:12:0,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson.h:8,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:13,
                 from DroneGY80Final.ino:10:
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/ReferenceType.hpp: In copy constructor 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)':
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/Internals/ReferenceType.hpp:28:3: error: 'ArduinoJson::Internals::ReferenceType::ReferenceType(const ArduinoJson::Internals::ReferenceType&)' is private
   ReferenceType(const ReferenceType&);
   ^
In file included from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson.h:9:0,
                 from C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson.h:13,
                 from DroneGY80Final.ino:10:
C:\Users\Aaron\Documents\Arduino\libraries\ArduinoJson-master\src/../include/ArduinoJson/JsonObject.hpp:33:8: error: within this context
  class JsonObject : public Internals::JsonPrintable<JsonObject>,
        ^
DroneGY80Final.ino: In function 'void loopWifi()':
DroneGY80Final.ino:627:44: note: synthesized method 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)' first required here

bblanchon, 2015-09-22:
Hi Aaron,

I downloaded the Edison's core library with Arduino's "Boards Manager".

There is something strange in the `WString.h`.
Here is a comparison of original Arduino's AVR core (on the left) and Edison's (on the right):
https://www.diffchecker.com/pzhscm6k

As you can see `c_str()` is missing at line 163.

I don't know what happened to `WString.h`.
I think you should open an issue on Edison's issue tracker.

Regards,
Benoit

creativeknights2012, 2015-09-22:
Will do, and I'll get back to you on the outcome.

creativeknights2012, 2015-09-22:
I added in the missing lines to the WString.h file, and now it works. Thanks.

closed 2015-09-22 by creativeknights2012

bblanchon, 2015-09-22:
Did you report the problem to Intel or Arduino?

creativeknights2012, 2015-09-22:
Unfortunately not. I will though.

bblanchon, 2015-09-22:
Thanks, I also made a PR: https://github.com/01org/corelibs-edison/pull/18
I hope it's in the right place.

creativeknights2012, 2015-09-24:
Hi, I just wanted to follow up on this issue with another possible resolution as well. 
This article states that by replacing the c_str() converter command with system(String.buffer).

http://www.costaricamakers.com/?p=190

bblanchon, 2015-09-24:
Using `c_str()` is a better solution:

1. `c_str()` is compatible with [original's Arduino `String`](https://www.arduino.cc/en/Reference/CStr)
2. `c_str()` is compatible with the [STL](http://www.cplusplus.com/reference/string/string/c_str/)
3. `buffer` relies on a public field that is [very likely to be hidden](https://google-styleguide.googlecode.com/svn/trunk/cppguide.html#Access_Control) in a future release

So I would say that using `String::buffer` is a pretty bad advice.

cruzmendezalan, 2016-05-28:
One solution is to look for the "WString.h" file and add  in  "character access" 

 const char * c_str() const { return buffer; }

ObjSal, 2017-01-03:
They've fixed this in trunk, we just need to wait for them to push it to the boards manager, I'm using the trunk version right now. https://github.com/01org/corelibs-edison/pull/18