bblanchon/ArduinoJson #125: ESP8266 crashes when retrieving values after successful parsing

gvuyk, 2015-09-23:
Example:
```c++
#include <ArduinoJson.h>

void setup()
{
  Serial.begin(115200);

  String json_string = "{\"value\": 3}";
  StaticJsonBuffer<200> jsonBuffer_im;
  JsonObject& json_obj = jsonBuffer_im.parseObject(json_string);
  
  if (!json_obj.success())
  {
    Serial.println("Error parsing json");
    return;
  }
  
  int value = json_obj["value"];
  Serial.println(value);
}

void loop(){}
```
ESP8266 output:
```
Exception (29):
epc1=0x4020424f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffe9b40 end: 3ffe9e10 offset: 01a0

>>>stack>>>
3ffe9ce0:  3fffdc20 3ffe8cfc 3fff3888 3ffe9e3c  
3ffe9cf0:  3fffdc20 3ffe8cfc 3ffe9d1a 40202688  
3ffe9d00:  3fffdc20 3ffe8cfc 3fff3888 4020216b  
3ffe9d10:  3ffe8850 756c6176 00330065 7d33203a  
3ffe9d20:  00000000 3ffe9d10 3ffe9d2c 00000000  
3ffe9d30:  00000000 3ffe9d14 00000000 00000001  
3ffe9d40:  0000000c 3ffe9d1a 40204a1a 00000000  
3ffe9d50:  00000000 00000000 00000000 00000000  
3ffe9d60:  00000000 00000000 00000000 00000000  
3ffe9d70:  00000000 00000000 00000000 00000000  
3ffe9d80:  00000000 00000000 00000000 00000000  
3ffe9d90:  00000000 00000000 00000000 00000000  
3ffe9da0:  00000000 00000000 00000000 00000000  
3ffe9db0:  00000000 00000000 00000000 00000000  
3ffe9dc0:  00000000 00000000 00000000 00000000  
3ffe9dd0:  00000000 00000000 00000000 00000038  
3ffe9de0:  3fff3888 0000000c 0000000c 00000000  
3ffe9df0:  00000000 00000000 3ffe9e34 40201c63  
3ffe9e00:  00000000 00000000 3ffe8df0 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```
Workaround: 
```c++
  String value_str = json_obj["value"].asString();
  int value = value_str.toInt();
  Serial.println(value);
```

bblanchon, 2015-09-23:
Probably another alignment issue.
I received my ESP8266, I'll try to have a look at it the week end.

bblanchon, 2015-09-26:
I didn't check yet, but there is a good chance that this is issue #101.
It was a bug in ESP8266's core, but I think they fixed it in commit [47b7bcb](https://github.com/esp8266/Arduino/commit/47b7bcbeb5d8640205a4d9c2acd8b3817fd3caaa)

Please check that the Arduino core for ESP8266 is up to date on your machine.
Otherwise please try the fixes proposed in #101.

gvuyk, 2015-09-26:
Yep, the fix is on *staging*. Thanks

closed 2015-09-26 by gvuyk