bblanchon/ArduinoJson #126: Help with project

rahul2704, 2015-09-27:
Hey there. this is a very nice project. I have an array of values which I am sending from browser.
I use the method JSON.stringify before sending message. the arduino sketch running server reads it as string, e.g {"x":"0","y":"150","z":"99"}, how do I parse this into json to extract values of x, y and z in different variables? thanks
 I tried the Json parser example to implement my sketch, 
here is the code 

    case WStype_TEXT:
        Serial1.printf("[%u] get Text: %s\n", num, payload);
        { String text = String((char *) &payload[0]);
             
           StaticJsonBuffer<200> jsonBuffer;

         char json[] = text;

      JsonObject& root = jsonBuffer.parseObject(json);

        if (!root.success()) {
        Serial.println("parseObject() failed");
      return;
      }

      int x = root["x"];
      int y = root["y"];
      int z = root["z"];
      }
                webSocket.sendTXT(num, payload, lenght);
                webSocket.broadcastTXT(payload, lenght);
                break;`

i keep getting this error 

    get_payload.ino: In function 'void webSocketEvent(uint8_t, WStype_t, uint8_t*, size_t)':
    get_payload:43: error: initializer fails to determine size of 'json'
    get_payload:43: error: array must be initialized with a brace-enclosed initializer
    initializer fails to determine size of 'json'

      This report would have more information with
      "Show verbose output during compilation"
      enabled in File > Preferences.
   

in your example, you already have the json defined, but how to define it from the values I send from browser? thank you

rahul2704, 2015-09-27:
I tried replacing 'json' in code directly with text

    JsonObject& root = jsonBuffer.parseObject(json); 
to this
    JsonObject& root = jsonBuffer.parseObject(text);
should it work?

bblanchon, 2015-09-27:
Yes, it should work, but the best would be to use the `payload` buffer directly:

```c++
JsonObject& root = jsonBuffer.parseObject(payload);
```

Please don't use `String`, where did you get that idea anyway?

rahul2704, 2015-09-27:
thanks for the quick reply. because when I use payload direcrly it throws overload error etc. also i tried debugging. 
first I sent JSON from webbrowser directly, mcu recieves object
secondly i sent JSON.stringify, i can see the string in serial monitor
BUT, in both cases,  I get parseObject() failed. even by feeding payload directly. I tried a lot of permutations, no success. can you help?

bblanchon, 2015-09-27:
Well, there is very little material in your message...

Please try to give more detailed information.
Can you provide a **short** but **complete** program that shows the problem?

Also please [read the manual](https://github.com/bblanchon/ArduinoJson/wiki).

rahul2704, 2015-09-27:
i am running a similar sketch on esp8266 mcu https://github.com/Links2004/arduinoWebSockets/blob/master/examples/WebSocketServer/WebSocketServer.ino

It runs websocket server. i was successful in sending values from webbrowser. single values are easy to catch and use in code. but i want to send acccelerometer values as  a JSON, so i need to use your code to  parse the json received from webbrowser to further do some things in microcontroller. I am using .send(msg of websocket method in javascript. and yeah, i read the manual. but im sorry, I am a beginner at coding, and cannot understand a lot. Im learning as I go. thanks for your patience

bblanchon, 2015-09-28:
Please try:

    JsonObject& root = jsonBuffer.parseObject((char*)payload);

If you still have a compilation error, please copy/paste the message.

BTW, it's fine being a beginner; i'm just saying that you'd get quicker and better responses if you're very specific about your problem.

closed 2015-10-06 by bblanchon

felangga, 2017-05-30:
sketch_may25a:36: error: crosses initialization of 'ArduinoJson::JsonObject& root'
       JsonObject& root = jsonBuffer.parseObject((char*)payload);