bblanchon/ArduinoJson #127: Send out data to wifi.client without copying first..

sticilface, 2015-09-29:
I'm working to implement phillips Hue and the ESP8266.  The example i've picked up uses this to send out the JSON
```
void sendJson(aJsonObject *root)
{
  // Take aJsonObject and print it to Serial and to WiFi
  // From https://github.com/pubnub/msp430f5529/blob/master/msp430f5529.ino
  char *msgStr = aJson.print(root);
  aJson.deleteItem(root);
  HTTP.send(200, "text/plain", msgStr);
  free(msgStr);
}
```
However, this copies the string first.  I'm fairly sure you can write it straight to the client with client.write or something like that?  Am i correct, or is there a reason why not to?  I'm quite new to this... the Hue stuff is fairly complex.  I'm making progress...slowly..

The ESP will print back a response to the client that is already.. connected.. but I'm struggling with the exact semantics of the networking stuff / HTTPserver implementation of the ESP.  

```
void writeResponse(EthernetClient& client, JsonObject& json) {
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  client.println();
  json.prettyPrintTo(client);
}
```

bblanchon, 2015-09-29:
It's certainly possible to do that with ArduinoJson.

I cannot find the definition of `HTTPserver`, where is it defined?
I probably contains a member of type `EthernetClient` that you can use as a target for `JsonObject::printTo()`.

sticilface, 2015-09-29:
I might be making progress.  thanks for the fast response.  Some time to stew it over might have helped. 

```
 void writeResponse(JsonObject& json) {
  WiFiClient client = HTTP.client();
   client.println("HTTP/1.1 200 OK");
   client.println("Content-Type: application/json");
   client.println("Connection: close");
   client.println(); 
   json.prettyPrintTo(client);
 }
``` 
It will take me a while to get it all together, as I'm moving from aJSON to arduinoJson..

sticilface, 2015-10-06:
OK, problem solved.  I ended up writing my own HTTP buffer, that can determines content-length and manually constructing the json as the resulting file can be 65K, with a memory overhead of only 1K.

closed 2015-10-06 by sticilface

rmhomecouk, 2015-10-15:
@sticilface would you be willing to share your code please?

sticilface, 2015-10-15:
sure can.  its already on github. 
https://github.com/sticilface/Esp8266-Hue

its the HTTPprinter.h + .cpp

examples of how it is used can be found in the HUEbridge stuff...

rmhomecouk, 2015-10-15:
@sticilface - Wonderful, thanks :+1:

sticilface, 2015-10-15:
no problem.  its a bit rough round the edges and needs some features removed for the sake of clutter.  if ur struggling to use it post an issue.. as I'm sure you will not be the last..  its an odd way of doing things, that i've not seen before... but it seems to work..reliably..