bblanchon/ArduinoJson #128: Create function to add new items to array

sticilface, 2015-09-29:
I seem to be struggling a little, forgive me for being a noob. 

I want to create a function to add items to an array, one that is root.  I've written this function
```
void AddSucessToArray(JsonArray& array, String item, String value) {
      JsonObject& nestedObject = array.createNestedObject(); 
      JsonObject& sucess = nestedObject.createNestedObject("sucess");
      sucess[item.c_str()] = value.c_str(); 
}
```

but each time I call it using say (4 times) 
```
    AddSucessToArray (array, response, String(sat)); 
```

I get an array populated with the right number items, one for each time I call the function.  But all the values are the same. 
```
[
  {
    "sucess": {
      "/lights/1/state/bri": "151"
    }
  },
  {
    "sucess": {
      "/lights/1/state/bri": "151"
    }
  },
  {
    "sucess": {
      "/lights/1/state/bri": "151"
    }
  },
  {
    "sucess": {
      "/lights/1/state/bri": "151"
    }
  }
]
```
This probably has something to do with the passing by reference, but I'm just not sure which bit is causing it.

bblanchon, 2015-09-29:
You are calling `c_str()` on temporary `String`.

This should fix the issue:

    sucess[item] = value;

This is also explained in the manual:
https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#6-do-not-assume-that-strings-are-copied

sticilface, 2015-09-29:
That sorted it out.  Thanks.  I did read your common mistake page, but failed to make the connection with c_str().

closed 2015-09-29 by sticilface