bblanchon/ArduinoJson #129: Weird exception with certain son constructions

sticilface, 2015-09-30:
Again sorry to post, got a weird error... 

This code 
```
void Get_Fullstate() {
    StaticJsonBuffer<2000> jsonBufferOUT; 
    JsonObject& object = jsonBufferOUT.createObject();
    JsonObject& lightsLIST = object.createNestedObject("lights");
    JsonObject& GroupsLIST = object.createNestedObject("groups" ); 
     JsonObject& ScenesLIST = object.createNestedObject("scenes" ); 
     JsonObject& ConfigLISTb = object.createNestedObject("configX"); 
    object.prettyPrintTo(Serial);   
}
```
generates this JSON
```
{
  "lights": {},
  "groups": {},
  "scenes": {},
  "configX": {}
}
```

This code with the "configX" changed to "config" 

```
void Get_Fullstate() {
    StaticJsonBuffer<2000> jsonBufferOUT; 
    JsonObject& object = jsonBufferOUT.createObject();
    JsonObject& lightsLIST = object.createNestedObject("lights");
    JsonObject& GroupsLIST = object.createNestedObject("groups" ); 
     JsonObject& ScenesLIST = object.createNestedObject("scenes" ); 
     JsonObject& ConfigLISTb = object.createNestedObject("config"); 
    object.prettyPrintTo(Serial);   
}
```
causes this
```
{
  "lights": {},
  "groups": {},
  "scenes": {},
  "config": {}
}
Exception (9):
epc1=0x40101bfd epc2=0x00000000 epc3=0x00000000 excvaddr=0x666e6f63 depc=0x00000000

ctx: cont 
sp: 3ffebe10 end: 3ffec080 offset: 01a0

>>>stack>>>
3ffebfb0:  4020154e 00000001 00000001 40201543  
3ffebfc0:  3fff3450 000007d0 3ffea068 402073f9  
3ffebfd0:  00000000 00000000 3fff3b88 00000008  
3ffebfe0:  00000008 3fff3b70 00000003 00000003  
3ffebff0:  3fff3f88 00000082 00000000 3ffeafb8  
3ffec000:  3fffdc20 3ffec0ac 40201cb4 0000000a  
3ffec010:  4020154e 00000001 00000001 3ffec0ac  
3ffec020:  3fffdc20 000003d7 3ffea068 40207465  
3ffec030:  3ffe9240 00000000 000003e8 00003943  
3ffec040:  3ffea080 3fff3e50 00000016 40102441  
3ffec050:  40201cd5 00000000 3ffec0a4 402024a2  
3ffec060:  3fffdc20 00000000 3ffec0a4 40201d32  
3ffec070:  00000000 00000000 3ffeb060 40100528  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
'''

sticilface, 2015-09-30:
ok, if i reduce the buffer to 500 it works.

bblanchon, 2015-10-01:
I think the default a stack size is 4KB on the ESP8266.
If you need a bigger `JsonBuffer`, you should switch to `DynamicJsonBuffer` because it makes allocation on the heap.

But you should also question yourself on why you would need such a big buffer...

asetyde, 2015-10-01:
I use dynamic , at now <= 505 bytes is limit supported after crash with stack error

sticilface, 2015-10-01:
Ah, I switched to dynamic and it is working fine, so thanks.  
The reason is that I'm working with phillips hue, and there is a command that is issued by clients that requires the whole config and light states lot to be dumped into one json... this gets massive.

closed 2015-10-01 by sticilface