bblanchon/ArduinoJson #130: problem in sending data serially when i simultaneously try to read serial data.

abhaysbharadwaj, 2015-10-01:
hello,
i just started using the ArduinoJson library for my project. my aim is to send three sensor values using BT to a RasPi from my arduino nano using json format. Also, control a relay by receiving '0' or '1' from BT.

I am successfully able to send data. But, when i include the following piece of code: (to read serial data)
if (Serial.available > 0)
{
     z = Serial.read();
      ...............
}

and i send a '1'; my data transmission is disturbed and the json string is affected. when i send '0', it goes back to normal.

i need help in solving this issue as im unable to find out where i am going wrong!! kindly do the needful!!

pictures of the serial monitor:

[image]
[image]
[image]

my code:

#include <ArduinoJson.h>
#include <SoftwareSerial.h>
#include <DHT.h>
#define DHTPIN 12     // what pin we're connected to
#define DHTTYPE DHT11   // DHT 11 sensor
DHT dht(DHTPIN, DHTTYPE); //initilize the DHT sensor
int ldr = A0;   // LDR connected to A0
int relay = 13;
float h,t;
int rx = 11; // softwareserial rx of arduino is pin 11
int tx = 10; // softwareserial tx of arduino is pin 12
SoftwareSerial mySerial(rx,tx);

//int condition, value; //relay control parameters
int l;  // variable to store the value coming from the sensor
int bt = 0; // variable to store incoming BT data

void setup() 
{
  pinMode(relay, OUTPUT);
  pinMode(ldr, INPUT);
  Serial.begin(9600);
  mySerial.begin(9600);
  mySerial.println(" Project FLIP");
  mySerial.println("sensor testing");
  mySerial.println(" DHT11 test!");
  dht.begin();
  delay(2000); 
}

void loop() 
{
  humidity_read();//read humidity value
  temperature_read();//read temperature
  light_read(); //read LDR value
  jason_print();
  relay_control();//control relay by reading serial data 0=off, 1=on
  jason_print();
}

void jason_print()
{
  StaticJsonBuffer<200> temp;
  JsonObject& root = temp.createObject();
  root["humidity"] = h;
  root["temp"] = t;
  root["light"] = l;
  root.printTo(Serial);
  Serial.println();
  delay(1000);
}

void humidity_read()
{
  h = dht.readHumidity();  // Reading temperature or humidity takes about 250 milliseconds!
}

void temperature_read()
{
  t = dht.readTemperature();  // Read temperature as Celsius (the default)
}

void relay_control()
{
  if (Serial.available() > 0)
  {
    bt = Serial.read();
    if (bt == '1')
    {
      digitalWrite(relay, HIGH);
    }
    else if (bt == '0')
    {
      digitalWrite(relay, LOW);
    }
  }
}

void light_read()
{
  l = analogRead(ldr); //Read LDR value
}

bblanchon, 2015-10-01:
I'm sorry but here is not the right place to ask that kind of question.
I quickly looked at your code, but the answer is not obvious and it probably isn't caused by ArduinoJson.

Please post your question to [stackoverflow.com](http://stackoverflow.com/), you'll reach a wider audience.

Here it's just me, and I already have enough work with ArduinoJson.
But please come back if there is a problem coming from the library itself.

closed 2015-10-01 by bblanchon

abhaysbharadwaj, 2015-10-01:
ok thanks!!