bblanchon/ArduinoJson #131: Howto remove keys from JsonObject

murmlgrmpf, 2015-10-04:
Hi Benoit Blanchon,
Thank you for the great library! I am using it to provide an interface to a heating system.
Is there a simple way to remove a key value pair from a JsonObject containing multiple keys?
Alternatively, is it possible to generate a json string containing just one key or a subset of the keys.

I was trying 
obj["key"].printTo(Serial);

But it will just send the value not the json string.

Thanks,
Thomas

bblanchon, 2015-10-04:
Hi Thomas,

[JsonObject::remove()](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/JsonObject.hpp#L115) is what you're looking for:

```c++
obj.remove("key");
```

Regards,
Benoit

murmlgrmpf, 2015-10-05:
Thank you! I was sure that there should be a remove command, but i couldn't find it in the documentation nor in the code.

closed 2015-10-05 by murmlgrmpf