bblanchon/ArduinoJson #132: Can't parse nested JSON on the ESP8266

vshymanskyy, 2015-10-05:
Just came across the same issue:
http://www.esp8266.com/viewtopic.php?p=21821

bblanchon, 2015-10-05:
The buffer is too small:

1. root object's size = `JSON_OBJECT_SIZE(2)` = 72 bytes
2. first nested object's size = `JSON_OBJECT_SIZE(2)` = 72 bytes
3. second nested object's size = `JSON_OBJECT_SIZE(2)` = 72 bytes

Total = 216 bytes.

You need to increase the size of the `StaticJsonBuffer` or switch to `DynamicJsonBuffer`.

domonetic, 2015-10-05:
What happens , if I get something bigger than the buffer?

vshymanskyy, 2015-10-05:
It appears to be a *similar* issue. but actually i tracked it down to being not able to parse numbers on this platform. I have fixed it, I'll prepare a PR for that.
Thanks!

asetyde, 2015-10-05:
Actually max 505bytes work ...

closed 2015-10-06 by bblanchon