bblanchon/ArduinoJson #135: possible memory leak

grahamehorner, 2015-10-14:
reading and writing JsonObject using the StaticJsonBuffer<800> jsonBuffer; results in corrupt json
this may be due to an incorrect terminated string buffer.

bblanchon, 2015-10-14:
Would you mind elaborating a little?

grahamehorner, 2015-10-14:
I have create a small config.json method that has a number of vaules and nested JsonObjects; when writing out the root JsonObject to a file and reading back the file the data written has been appened with corruption.

bblanchon, 2015-10-14:
Please copy/paste the code here.

grahamehorner, 2015-10-14:
/**/
JsonObject& makeConfiguration(JsonBuffer& jsonBuffer)
{
  JsonObject& root = jsonBuffer.createObject();
  root.set("chip-id",ESP.getChipId());
  root.set("flash-id",ESP.getFlashChipId());

  JsonObject& module = jsonBuffer.createObject();
  module.set("type"   , MODULE_TYPE);
  module.set("version", MODULE_VERSION);
  module.set("release", MODULE_RELEASE);
  root.set("module-config", module);
    
  JsonObject& wifi = jsonBuffer.createObject();
  wifi.set("mode" , "STA");
  
  JsonObject& wifiAP = jsonBuffer.createObject();
  wifi.set("AP", wifiAP);
   
  JsonObject& wifiST = jsonBuffer.createObject();
  wifi.set("ST", wifiST);
  
  JsonArray& wifiAccessPointCollection = wifi.createNestedArray("access-points");
  int found = WiFi.scanNetworks();
  for(int network = 0; network < found; network++)
    wifiAccessPointCollection.add( WiFi.SSID( network ) );
 
  root.set("network-config" , wifi ); 
    
  return root;
}

/**/
JsonObject& loadConfiguration(String path, JsonBuffer& jsonBuffer)
{ 
  File jsonConfig = SPIFFS.open( path, "r");
  size_t size = jsonConfig.size();  
  
  std::unique_ptr<char[]> buf(new char[size]);
  jsonConfig.readBytes(buf.get(), size);
  
  DEBUG.println( buf.get() );
  return jsonBuffer.parseObject(buf.get());
}
/**/
bool saveConfiguration(String path, JsonObject& jsonObject)
{ 
  jsonObject.printTo(DEBUG);
  
  File jsonConfig = SPIFFS.open(path , "w");
  if (jsonConfig)
    jsonObject.printTo(jsonConfig);
  return jsonConfig;
}

bblanchon, 2015-10-14:
I assume that you're using the same instance of `StaticJsonBuffer` to call `makeConfiguration()` and `loadConfiguration()`.

This is probably the source of the problem.
Please read [Avoiding pitfall: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer).

In your case, a simple solution would be to declare two local `StaticJsonBuffer`s in `makeConfiguration()` and `loadConfiguration()`.

bblanchon, 2015-10-14:
PS: the suggested solution would not work because you're returning references to `JsonObject`.
You'll need to find another way to avoid reusing the same `JsonBuffer`.

grahamehorner, 2015-10-14:
@bblanchon I am indeed reusing the StaticJsonBuffer to avoid memory fragmentation; when loading/saving the configuration; how would you suggest I avoid memory fragmentation other then have a global StaticJsonBuffer?

bblanchon, 2015-10-14:
Just use a local or scoped `StaticJsonBuffer` :-)

`StaticJsonBuffer` is a [memory pool](https://en.wikipedia.org/wiki/Memory_pool) specially designed to avoid memory fragmentation.

You may want to read [this blob post](http://blog.benoitblanchon.fr/arduino-json-parser/).
Class names have changed since then, but the principle is exactly the same.

closed 2015-10-21 by bblanchon