bblanchon/ArduinoJson #138: ArduinoJon

prudencer, 2015-10-20:


bblanchon, 2015-10-20:
Did you check issue #123 ?

prudencer, 2015-10-20:
Yes I did, the Wstring.h file has been updated to include those missing lines of code, so line 163 with c_string is there. Im using Arduino 1.6.5

bblanchon, 2015-10-20:
Can you please copy and paste the output of the compiler?

prudencer, 2015-10-20:
This is the compiler output I get:

In file included from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:10:0,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariant.hpp:16,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:14,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,
                 from JsonGeneratorExample.ino:7:
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp: In constructor 'ArduinoJson::JsonObjectKey::JsonObjectKey(const String&)':
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp:17:48: error: 'const class String' has no member named 'c_str'
   JsonObjectKey(const String& key) : _data(key.c_str()) {}
                                                ^
In file included from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9:0,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,
                 from C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,
                 from JsonGeneratorExample.ino:7:
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonArray& ArduinoJson::JsonBuffer::parseArray(const String&, uint8_t)':
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:70:28: error: 'const class String' has no member named 'c_str'
     return parseArray(json.c_str(), nesting);
                            ^
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'ArduinoJson::JsonObject& ArduinoJson::JsonBuffer::parseObject(const String&, uint8_t)':
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:93:29: error: 'const class String' has no member named 'c_str'
     return parseObject(json.c_str(), nesting);
                             ^
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function 'char* ArduinoJson::JsonBuffer::strdup(const String&)':
C:\Users\Prudence\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:100:55: error: 'const class String' has no member named 'c_str'
   char *strdup(const String &src) { return strdup(src.c_str(), src.length()); }
                                                       ^
Error compiling.

bblanchon, 2015-10-21:
You probably patched the wrong `WString.h`, you need to modify this one:

```
C:\Users\Prudence\AppData\Roaming\Arduino15\packages\Intel\hardware\i686\1.6.2+1.0\cores\arduino\WString.h
```

closed 2015-10-27 by bblanchon