bblanchon/ArduinoJson #139: undefined reference to `ArduinoJson::Internals::Encoding::_escapeTable' and similar issues

edgar-ch, 2015-10-20:
I'm trying to built JsonGeneratorExample.ino using Stino plugin for Sublime Text 3 (use arduino 1.6.5 sources) and have some errors:
```
[ 91%] Creating /tmp/Stino_build/JsonGeneratorExample/core.a...
[ 94%] Creating /tmp/Stino_build/JsonGeneratorExample/JsonGeneratorExample.elf...
/tmp/Stino_build/JsonGeneratorExample/JsonGeneratorExample.ino.cpp.o: In function `setup':
/home/edgar/sketchbook/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:9: undefined reference to `ArduinoJson::Internals::List<ArduinoJson::JsonVariant>::addNewNode()'
/home/edgar/sketchbook/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:39: undefined reference to `vtable for ArduinoJson::Internals::IndentedPrint'
/home/edgar/sketchbook/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:39: undefined reference to `vtable for ArduinoJson::Internals::IndentedPrint'
/home/edgar/sketchbook/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:39: undefined reference to `vtable for ArduinoJson::Internals::Prettyfier'
/home/edgar/sketchbook/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:39: undefined reference to `vtable for ArduinoJson::Internals::Prettyfier'
/tmp/Stino_build/JsonGeneratorExample/core.a(JsonObject.cpp.o): In function `_GLOBAL__sub_I__ZN11ArduinoJson10JsonObject8_invalidE':
/home/edgar/sketchbook/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/List.hpp:32: undefined reference to `ArduinoJson::Internals::Encoding::_escapeTable'
/home/edgar/sketchbook/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/List.hpp:32: undefined reference to `ArduinoJson::Internals::Encoding::_escapeTable'
/home/edgar/sketchbook/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/List.hpp:32: undefined reference to `ArduinoJson::Internals::List<ArduinoJson::JsonPair>::addNewNode()'
collect2: error: ld returned 1 exit status
```
It seems like error in methods definition or something like this.

bblanchon, 2015-10-20:
Stino used to have an issue with the new library layout, see issue #74.
I submitted a [pull request](https://github.com/Robot-Will/Stino/pull/273) which has been merged, but i'm not sure it's available in the master branch.

Can you please try with the "old-layout" package ?

* ~~[ArduinoJson-v5.0.4.zip](https://github.com/bblanchon/ArduinoJson/releases)~~
* [ArduinoJson-v5.0.4-old-layout.zip](https://github.com/bblanchon/ArduinoJson/releases)

edgar-ch, 2015-10-20:
Now build without errors, thank you !

closed 2015-10-20 by bblanchon