bblanchon/ArduinoJson #140: Decoding complex nested object

dosary, 2015-10-21:
I'm still new to this library I was not sure how you can decode this code mentioned in Example1

```javascript
[
  {
    "error": {
      "address": "",
      "description": "link button not pressed",
      "type": "101"
    }
  }
]
```

I'm actually trying to decode this:
```javascript
{
    "IsSuccessful": false,
    "Error": {
        "Code": 23009,
        "Message": "Not mathced!"
    }
}
```
But I'm not sure how that's done.

I think it would be helpful for new people like me, thanks !

bblanchon, 2015-10-21:
```c++
JsonObject& root = jsonBuffer.parseObject(json);

bool isSuccessful   = root["IsSuccessful"];
int errorCode       = root["Error"]["Code"];
const char* message = root["Error"]["Message"];
```

dosary, 2015-10-22:
Thanks ! works like a charm!

closed 2015-10-22 by dosary