bblanchon/ArduinoJson #142: Need help with compile error.

towler73, 2015-10-29:
I'm trying to compile a sketch for [Lightblue Bean](http://legacy.punchthrough.com) and I'm getting compile error I need help diagnosing.  See sketch and error below.

Thanks,
Brett

Sketch
```
#include <ArduinoJson.h>

static int8_t temp = 0;

void setup() {
  Serial.begin(57600);
}

void loop() {
  int8_t newTemp = Bean.getTemperature();

  if(newTemp != temp) {
    
    StaticJsonBuffer<200> jsonBuffer;

    JsonObject& root = jsonBuffer.createObject();
    root["oldtemp"] = temp;
    root["temp"] = newTemp;  
    
    root.printTo(Serial);

    temp = newTemp;
  }
  Bean.sleep(1000);
}
```

Error
```
/Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/JsonVariant.cpp: In member function 'T ArduinoJson::JsonVariant::as() const [with T = String]':
/Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/JsonVariant.cpp:67:46: error: call of overloaded 'String(const double&, uint8_t&)' is ambiguous
     return String(_content.asDouble, decimals);
                                              ^
/Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/JsonVariant.cpp:67:46: note: candidates are:
In file included from /Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/Print.h:26:0,
                 from /Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Arduino/Print.hpp:29,
                 from /Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/DummyPrint.hpp:9,
                 from /Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/JsonPrintable.hpp:9,
                 from /Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/JsonVariant.hpp:12,
                 from /Users/bmorgan/Documents/Arduino/libraries/ArduinoJson/src/JsonVariant.cpp:7:
/Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/WString.h:70:11: note: String::String(long unsigned int, unsigned char)
  explicit String(unsigned long, unsigned char base=10);
           ^
/Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/WString.h:69:11: note: String::String(long int, unsigned char)
  explicit String(long, unsigned char base=10);
           ^
/Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/WString.h:68:11: note: String::String(unsigned int, unsigned char)
  explicit String(unsigned int, unsigned char base=10);
           ^
/Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/WString.h:67:11: note: String::String(int, unsigned char)
  explicit String(int, unsigned char base=10);
           ^
/Applications/Arduino.app/Contents/Java/hardware/LightBlue-Bean/avr/cores/bean/WString.h:66:11: note: String::String(unsigned char, unsigned char)
  explicit String(unsigned char, unsigned char base=10);
           ^
Error compiling.
```

bblanchon, 2015-10-29:
Hi Brett,

I looks like the LightBlue Bean core lacks a few things in its `String` class.
See here the differences: https://www.diffchecker.com/0s9u6xy1
On the left is the original Arduino `WString.h`, on the right is the one from LightBlue.

This explains the compiler error `call of overloaded 'String(const double&, uint8_t&)' is ambiguous`

There are two solutions:

1. The quick and dirty fix:
On your local copy remove the implementation of `String JsonVariant::as<String>()` from `JsonVariant.cpp`

2. The right thing to do:
Open an issue (or better a pull request) to LightBlue team.
https://github.com/PunchThrough/PunchThrough-BEAN-Arduino-Firmware

Please keep me informed with the progress.
Feel free to edit the [Compatibility issue on the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility%20issues).

Regards,
Benoit

towler73, 2015-10-29:
Benoit,

Thanks for your quick response.  I'll open an issue with LightBlue team and perhaps take a crack at fixing.

Brett

bblanchon, 2015-11-05:
They don't really seem to care...

mplewis, 2015-11-05:
Hey @bblanchon, sorry about the delay. We're looking into fixing the issue and figuring out the best way to do so. Thank you for the detailed bug report!

bblanchon, 2015-11-25:
The usage of this constructor has been removed from ArduinoJson 5.0.7.

closed 2015-11-25 by bblanchon

towler73, 2015-12-01:
Great, I'll try it out.  Thanks.

towler73, 2016-01-04:
Sorry, meant reply awhile ago.  I tested, and all works now.  Thanks.