bblanchon/ArduinoJson #144: String duplication in jsonBuffer, release 5

hreintke, 2015-10-31:
LS,
I am integrating ArduinoJson into the Sming Framework.
Sming is a framework for creating native applications for ESP8266.

Currently we use ArduinoJson 4.2 with local String extensions.
Within the local extension we have issues with memory consumption/leakage (due to our own implementation).
As Release 5 has included String support I am looking at the possibility to use ArduinoJson without local extensions.

I have been searching thru documentation but was not able to find answer.
If I missed just point me to that and I will work from there.
I did see this line but not further indication and just want to be sure.
`Note: ArduinoJson 5.0 (currently in beta), make copies of String values but not char*.`

The question is about string duplication in the jsonbuffer.

In the construction :
```
{
    StaticJsonBuffer<200> jsonBuffer;

	JsonObject& json = jsonBuffer.createObject();
    int i = 4;
    while (i--) {
        String k = "testing" + String(i) +String("_key");
        String v = "value"+ String(i+5) +String("_val");
        json[k] = v;
    }
    json.printTo(Serial);
}
```
Which uses local instead of global variables for key and value.

Is this a valid application ?
In other words : Will both Key and Value be copied into the jsonBuffer and be kept after Stings k and v are out of scope ?

bblanchon, 2015-10-31:
Yes, that's exactly what will happen.

The key is duplicated in [JsonObject.ipp, line 194](https://github.com/bblanchon/ArduinoJson/blob/c6a4bfa88679c7edf7f72a14f4ed7577ac6e7a56/include/ArduinoJson/JsonObject.ipp#L194).
The value is duplicated in [JsonObject.ipp, line 204](https://github.com/bblanchon/ArduinoJson/blob/c6a4bfa88679c7edf7f72a14f4ed7577ac6e7a56/include/ArduinoJson/JsonObject.ipp#L204).

As a consequence, you probably need to increase the size of the `StaticJsonBuffer`.

By the way, the size of the [stack is limited to 4K on the ESP8266](https://github.com/esp8266/Arduino/issues/808#issuecomment-142011991), this can be a problem if you need to a big `JsonBuffer`.
There are two solutions:

1. allocate `StaticJsonBuffer` in the heap (via `new` or `make_unique`)
2. switch to `DynamicJsonBuffer`

hreintke, 2015-10-31:
Clear.
And thanks for the good implementation and quick answer.

closed 2015-10-31 by hreintke