bblanchon/ArduinoJson #145: Invalid type for array subscript

hackscribble, 2015-11-01:
I'm trying to compile the canonical example from the wiki for an Arduino Uno using Arduino IDE 1.6.5.  The compiler reports the following errors:

```
sketch_nov01b.ino: In function 'bool deserialize(SensorData&, const char*)':
sketch_nov01b:10: error: invalid types 'const char*[const char [5]]' for array subscript
sketch_nov01b:11: error: invalid types 'const char*[const char [5]]' for array subscript
sketch_nov01b:12: error: invalid types 'const char*[const char [6]]' for array subscript
invalid types 'const char*[const char [5]]' for array subscript
```

My program and include files are as follows.  (I moved the struct and typedef to an include file to work round the known Arduino issue with structs).

```
#include <ArduinoJson.h>
#include "mystruct.h"

#define SENSORDATA_JSON_SIZE (JSON_OBJECT_SIZE(3))

bool deserialize(SensorData& data, const char* json)
{
  StaticJsonBuffer<SENSORDATA_JSON_SIZE> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);
  data.name = json["name"];
  data.time = json["time"];
  data.value = json["value"];
  return root.success();
}

void setup()
{
  char json[] = "{\"name\":\"Ray\",\"time\":12345,\"value\":3.14159}";
  SensorData myData;
  deserialize(myData, json);
}

void loop()
{
  
}
```

```
struct sensorData
{
  const char* name;
  int time;
  float value;
};

typedef struct sensorData SensorData;
```

bblanchon, 2015-11-01:
Shame on me! I should have tested the code before hitting "Save" :-)

You just need to remove `const` on the input JSON.
That's because ArduinoJson need to modify the string to insert the null-terminator and replace escaped chars.

I update the example on the wiki.

bblanchon, 2015-11-01:
Also, it's `root[something]` and not `json[something]`.
Sorry about that too.

hackscribble, 2015-11-01:
Thank you - working now :-)

closed 2015-11-01 by hackscribble

domonetic, 2015-11-02:
Sorry but in arduino 1.6.5 r5 with v5.0.5 The example compile only with this changes. Adding struct on parameters

#define SENSORDATA_JSON_SIZE (JSON_OBJECT_SIZE(3))

struct SensorData
{
   const char* name;
   int time;
   float value;
} ;

bool deserialize(struct SensorData& data, char* json)
{
    StaticJsonBuffer<SENSORDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(json);
    data.name = root["name"];
    data.time = root["time"];
    data.value = root["value"];
    return root.success();
}

void serialize(const struct SensorData& data, char* json, size_t maxSize)
{
    StaticJsonBuffer<SENSORDATA_JSON_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["name"] = data.name;
    root["time"] = data.time;
    root["value"] = data.value;
    root.printTo(json, maxSize);
}

bblanchon, 2015-11-02:
This is very strange.
This used to be like that in C, but I've never seen that with a C++ compiler.

BTW, the way to fix this is to change the definition of `SensorData`, like this:

```c++
typedef struct
{
    const char* name;
    int time;
    float value;
} SensorData;
```

domonetic, 2015-11-02:
yes, in other proyect run, but not in this

Arduino:1.6.5 (Windows 7), Placa:"Arduino/Genuino Uno"

dmtcesp8266:12: error: 'SensorData' was not declared in this scope
dmtcesp8266:12: error: 'data' was not declared in this scope
dmtcesp8266:12: error: expected primary-expression before 'char'
dmtcesp8266:12: error: expression list treated as compound expression in initializer [-fpermissive]
dmtcesp8266:13: error: variable or field 'serialize' declared void
dmtcesp8266:13: error: 'SensorData' was not declared in this scope
dmtcesp8266:13: error: 'data' was not declared in this scope
dmtcesp8266:13: error: expected primary-expression before 'char'
dmtcesp8266:13: error: expected primary-expression before 'maxSize'
dmtcesp8266.ino: In function 'bool deserialize(SensorData&, char*)':
dmtcesp8266:18: error: 'bool deserialize(SensorData&, char*)' redeclared as different kind of symbol
dmtcesp8266:12: error: previous declaration of 'bool deserialize'
'SensorData' was not declared in this scope

bblanchon, 2015-11-02:
I found the problem.

When Arduino preprocess the file, it **incorrectly** inserts function declarations before the `struct` definition.
Here is the sources file that has been generated on my `tmp` folder:

```c++
#include <ArduinoJson.h>
#include "Arduino.h"
bool deserialize(SensorData& data, char* json);
void serialize(const SensorData& data, char* json, size_t maxSize);
void setup();
void loop();
#line 9
typedef struct {
   const char* name;
   int time;
   float value;
} SensorData;

etc....
```

Indeed, `SensorData` is not declared when `deserialize()` is declared.
Adding `struct` acts as a forward declaration which fixes the problem.

But that just a workaround for this particular case. We need to find a more general solution or open an issue in Arduino's bug tracker.

bblanchon, 2015-11-02:
Here is a better workaround:

In Arduino IDE, create a new tab, call it `serialization.h` for instance.
Move the serialization code to the new file and include it from the main file.

Note that you need to keep the include of ArduinoJson in the main file (another Arduino IDE oddity).

domonetic, 2015-11-02:
in other project i use in separate file and work fine.
Thanks.

bblanchon, 2015-11-02:
Yes, that's what i was talking about.

bblanchon, 2015-11-09:
@domonetic I just saw that it's [explained in the Arduino Wiki](https://github.com/arduino/Arduino/wiki/Build-Process#pre-processing):

> The #include statement and function prototypes are inserted after any comments or pre-processor statements (#includes or #defines), but before any other statements (including type declarations). This means that if you want to use a custom type as a function argument, you should declare it within a separate header file.

domonetic, 2015-11-11:
what IDE do you use?