bblanchon/ArduinoJson #146: String and ArduinoJson::JsonObjectSubscript<const char*>

hreintke, 2015-11-02:
LS,

I think I have "just a C++" question, so if there is another location where I should ask the question just tell me.

I have successfully integrated ArduinoJson Release 5 into the Sming Framework.
Sming uses a Wiring WString implementation (adapted) for the String Class.

In the application I have the following construction.
```
	JsonObject& root = jsonBuffer.parseObject(jsonString);
	String ssid;
	
	JsonObject& network = root["network"];
	ssid = network["ssid"];
```

But then I get the compile error :
```
include/AppSettings.h:39:9: error: ambiguous overload for 'operator=' (operand types are 'String' and 'ArduinoJson::JsonObjectSubscript<const char*>')
    ssid = network["ssid"];
         ^
In file included from c:\userdata\sming_hr1022local_project\sming\wiring\wiringframeworkincludes.h:20:0,
                 from c:/userdata/Sming_HR1022Local_Project/Sming/SmingCore/SmingCore.h:13,
                 from app/application.cpp:2:
c:\userdata\sming_hr1022local_project\sming\wiring\WString.h:88:24: note: candidate: String& String::operator=(const String&)
     String & IRAM_ATTR operator = (const String &rhs);
                        ^
c:\userdata\sming_hr1022local_project\sming\wiring\WString.h:89:24: note: candidate: String& String::operator=(const char*)
     String & IRAM_ATTR operator = (const char *cstr);
                        ^
c:\userdata\sming_hr1022local_project\sming\wiring\WString.h:91:14: note: candidate: String& String::operator=(String&&)
     String & operator = (String && rval);
```
In order to get it work I need to use the following.
`ssid = String((const char*)network["ssid"]);`

As you are using a standard WString I expected something in the ArduinoJson library to make this work.
But I have been searching the library and examples but cannot find an indication.

Can you give me a hint for solving this ?

bblanchon, 2015-11-02:
It's issue #118.

I'm not sure I'll be able to fix this completely, but the following statement can be used instead:

```c++
ssid = (String)network["ssid"];
ssid = (const char*)network["ssid"];
ssid = network["ssid"].asString();
ssid = network["ssid"].as<const char*>();
ssid = network["ssid"].as<String>();
```

bblanchon, 2015-11-02:
Added to the FAQ:
https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-fix-error-ambiguous-overload-for-operator

hreintke, 2015-11-02:
Sorry, I indeed didn't check the issues as I was expecting it was an error from me.

FYI : In my environment the `ssid = (String)network["ssid"];` option also gives the 
`call of overloaded 'String(ArduinoJson::JsonObjectSubscript<const char*>)' is ambiguous` errormessage. But I have enough alternatives.

Thanks for the help.

closed 2015-11-02 by hreintke