bblanchon/ArduinoJson #147: Arduino IDE 1.6.6. warning ...

Apollon77, 2015-11-04:
Arduino IDE 1.6.6 brings this wanring on compile time:

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'

Seems to be that a small change in library.-property can fix this :-)

bblanchon, 2015-11-04:
I wonder which category I should use:

* Communication
* Data Storage
* Data Processing
* Other

They all seems to match more or less.

Apollon77, 2015-11-04:
I personally would see "Data Processing" the best fit ... because you parse JSon data ... and this is the basic functionality for later "storage" and/or "communication" ... My personal opinion ...

domonetic, 2015-11-04:
"Data Processing" +1

bblanchon, 2015-11-06:
Nevermind... i'll close the issue when this will be available in the library manager

bblanchon, 2015-11-09:
Fixed in version 5.0.6

closed 2015-11-09 by bblanchon