bblanchon/ArduinoJson #148: how to detect nested values?

domonetic, 2015-11-04:
I need to know if a value is an array, or other data.
What function use to detect it?

I want to receive this structure

{
     "command" WRITE,
     "function": OUTPUT
     "value" -> this can be string, integer or array;
}

example
{
     "command" WRITE,
     "function": OUTPUT
     "value" : {1:TRUE}
}

{
     "command" WRITE,
     "function": TIMER
     "value" : 1200
}

bblanchon, 2015-11-05:
```c++
JsonObject& root = jsonBuffer.parseObject();
JsonVariant value = root["value"];

if (value.is<const char*>()) {
    // it's a string
}

if (value.is<int>()) {
    // it's an integer
}

if (value.is<float>()) {
    // it's a float
}

if (value.is<JsonObject&>()) {
    // it's an object
}

if (value.is<JsonArray&>()) {
    // it's an array
}
```

domonetic, 2015-11-05:
I can help make a list of functions of the classes to add to the wiki?

bblanchon, 2015-11-05:
That's number one on my TODO list.
I want to write an "API reference" page.
Just give me a minute so I write a skeleton...

domonetic, 2015-11-05:
Yes please. I help you to add a description

bblanchon, 2015-11-05:
Here is the skeleton of what I had in mind:
https://github.com/bblanchon/ArduinoJson/wiki/API-Reference

domonetic, 2015-11-05:
Well, I started with some things I understand. And the tedious repeated blocks.

I keep going?

bblanchon, 2015-11-05:
It's excellent!
Yes, please go on.
:+1:

closed 2015-11-17 by bblanchon