bblanchon/ArduinoJson #149: no matching function for call to 'ArduinoJson::StaticJsonBuffer<200u>::parseObject(ArduinoJson::JsonObject&)'

f0ntana, 2015-11-05:
My code is:
```
Serial.println(response);

StaticJsonBuffer<200> jsonBuffer;

JsonObject& json = jsonBuffer.createObject();
JsonArray& data = json.createNestedArray("data");
data.add(response);
Serial.println("Json: ");
json.printTo(Serial);

JsonObject& root = jsonBuffer.parseObject(json);

```

this line don't work

```
JsonObject& root = jsonBuffer.parseObject(json);
```
give me this error:

no matching function for call to 'ArduinoJson::StaticJsonBuffer<200u>::parseObject(ArduinoJson::JsonObject&)'

bblanchon, 2015-11-05:
I'm sorry but what do you expect from this line?

    JsonObject& root = jsonBuffer.parseObject(json);

Maybe you wanted to write this instead?

    JsonObject& root = jsonBuffer.parseObject(response);

f0ntana, 2015-11-05:
This is my Serial when i use: ``` JsonObject& root = jsonBuffer.parseObject(response); ```

```
Aproxime o seu cartao do leitor...

UID da tag : 23 6A 82 C6
Reconhecendo cartao
connecting...

Id da Tag:  23 6a 82 c6

Id da Tag Sem Espaco: 236a82c6

connected
"{"success":true,"name":"Felipe Fontana"}"
Json: 
{"data":["\"{\"success\":true,\"name\":\"Felipe Fontana\"}\""]}parseObject() failed
```

bblanchon, 2015-11-05:
This is not valid JSON:

    "{"success":true,"name":"Felipe Fontana"}"

This is valid JSON:

    {"success":true,"name":"Felipe Fontana"}

You need to removed the extra quotes

f0ntana, 2015-11-05:
Now the code is:

```
Serial.println(response);

    StaticJsonBuffer<200> jsonBuffer;

    JsonObject& root = jsonBuffer.parseObject(response);
    
    if (!root.success())
    {
      Serial.println("parseObject() failed");
      return;
    }
    else{  
      Serial.println("parseObject() success");
      Serial.println("");
      return;
    }

    root.printTo(Serial);
    String name = root["name"];
    Serial.println(name);

```

Serial:
```
Aproxime o seu cartao do leitor...

UID da tag : 23 6A 82 C6
Reconhecendo cartao
connecting...

Id da Tag:  23 6a 82 c6

Id da Tag Sem Espaco: 236a82c6

connected
{"success":true,"name":"Felipe Fontana"}
parseObject() success
```

the API return:  ```{"success":true,"name":"Felipe Fontana"}```

this part don't work:
``` root.printTo(Serial);
    String name = root["name"];
    Serial.println(name);```

bblanchon, 2015-11-05:
Maybe because you wrote a `return` statement?

f0ntana, 2015-11-05:
Maybe u are right! Maybe i love u!
HAHAHAHA

Thanks a lot man!

bblanchon, 2015-11-05:
You're welcome.
Thanks for using the library.

closed 2015-11-05 by bblanchon