bblanchon/ArduinoJson #151: Issue with JSON_OBJECT_SIZE on ESP8266

marvinroger, 2015-11-06:
Here is a problematic code:

```c++
StaticJsonBuffer<JSON_OBJECT_SIZE(2)> json_buffer;
String json = String("{\"test1\":\"test\",\"test2\":\"test\"}");

// 1st attempt
JsonObject& parsed_json = json_buffer.parseObject(json); // parsed_json.success() -> false

// 2nd attempt
JsonObject& parsed_json = json_buffer.parseObject(json.c_str()); // parsed_json.success() -> false

// 3rd attempt
JsonObject& parsed_json = json_buffer.parseObject((char*)json.c_str()); // parsed_json.success() -> true
```
As explained [here](https://github.com/bblanchon/ArduinoJson/wiki/Decoding-JSON#step-2-parse-the-json-string):

> parseObject() takes a `char*` as a parameter. Be careful, it's not a `const char*`, the memory must be writable.

But as stated [here](https://github.com/bblanchon/ArduinoJson/blob/b9e3255c9e8680403dc0f7a409ddff4895c80ffe/include/ArduinoJson/JsonBuffer.hpp#L86), the function can also take a `String` or `const char*` but will make a copy of this string.

So all 3 should work, I guess... What's wrong?

bblanchon, 2015-11-07:
There is not enough room in `json_buffer` to store the copy of the string.
Therefore, only the overload that makes no copy can work.

To make them work, you need to increase the size of the `StatisJsonBuffer` of the size of the string (plus some bytes for the alignment and the terminator).

```c++
#define MAX_INPUT_LENGTH 40
StaticJsonBuffer<JSON_OBJECT_SIZE(2)+MAX_INPUT_LENGTH> json_buffer;
```

marvinroger, 2015-11-07:
Oh, okay thanks for the information. This would deserve to be in the Wiki, don't you think?
Feel free to close.

bblanchon, 2015-11-07:
I started an [API reference](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference) section, this will eventually go there.

closed 2015-11-07 by bblanchon