bblanchon/ArduinoJson #152: Initial size of DynamicJsonBuffer

bblanchon, 2015-11-07:
Add a `size_t` argument to the constructor that specifies the initial size of the buffer.

closed 2015-11-09 by bblanchon

marvinroger, 2016-03-22:
@bblanchon I'm trying to use this:

```c++
#include <ArduinoJson.h>

DynamicJsonBuffer generatedJsonBuffer;

void setup() {
  generatedJsonBuffer = new DynamicJsonBuffer(JSON_OBJECT_SIZE(3));
}

void loop() {
  // not used in this example
}
```

This fails with a

> error: invalid conversion from 'ArduinoJson::DynamicJsonBuffer* {aka ArduinoJson::Internals::BlockJsonBuffer<ArduinoJson::Internals::DefaultAllocator>*}' to 'size_t {aka unsigned int}'

I am probably using it in a bad way... Right?

marvinroger, 2016-03-22:
@bblanchon pinging you, please ignore my previous request.