bblanchon/ArduinoJson #153: OpenWeatherMap - very complex/multilevel Json data.

ethelder, 2015-11-07:
Hello,

I'm trying to work with API for weather forcasting and had problems parsing data about weather id from this api call (9 hours ahead) :

{"city":{"id":7530858,"name":"Poznań","coord":{"lon":16.901541,"lat":52.400631},"country":"PL","population":0,"sys":{"population":0}},"cod":"200","message":0.0154,"cnt":3,"list":[{"dt":1446919200,"main":{"temp":285.84,"temp_min":285.84,"temp_max":285.84,"pressure":1027.1,"sea_level":1036.19,"grnd_level":1027.1,"humidity":95,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10n"}],"clouds":{"all":92},"wind":{"speed":2.86,"deg":262.503},"rain":{"3h":0.045},"sys":{"pod":"n"},"dt_txt":"2015-11-07 18:00:00"},{"dt":1446930000,"main":{"temp":285.59,"temp_min":285.588,"temp_max":285.59,"pressure":1026.19,"sea_level":1035.37,"grnd_level":1026.19,"humidity":95,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10n"}],"clouds":{"all":56},"wind":{"speed":3.87,"deg":225.5},"rain":{"3h":0.0225},"sys":{"pod":"n"},"dt_txt":"2015-11-07 21:00:00"},{"dt":1446940800,"main":{"temp":286.55,"temp_min":286.549,"temp_max":286.55,"pressure":1024.19,"sea_level":1033.33,"grnd_level":1024.19,"humidity":95,"temp_kf":0},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10n"}],"clouds":{"all":80},"wind":{"speed":5.6,"deg":223.502},"rain":{"3h":0.01},"sys":{"pod":"n"},"dt_txt":"2015-11-08 00:00:00"}]}

I want to get all of the weather id's and than to find out - will it be raining:)
I know you have an example of "complex" in your wiki, but using JsonArray I still had problem parsing the value. I started a thread on Arduino forum as well.
https://forum.arduino.cc/index.php?topic=357973.new

BTW - your example doesn't work on Arduino Yun until you add a while(!Serial) loop in setup.

Thanks for any help in advance.

bblanchon, 2015-11-07:
Hi,

There are two pifalls with this kind of input:

1. You have to make the `JsonBuffer` large enough for to store the whole JSON tree.
2. You have to find the right path to the value in that tree.

To solve the first issue, you can either use a huge `StaticJsonBuffer` or a `DynamicJsonBuffer`. In both case you need to make sure that everything will fit into the RAM: the JSON input string, the `JsonBuffer` and all other global and local variables.
You'll know that this first step is OK, when `parseObject()` will succeed.

Then to extract the value, it seems that you just want to iterate through all the instance of `list[*].weather[0].id` which is done with the following code:

```c++
JsonObject& root = jsonBuffer.parseObject();
JsonArray& list = root["list"];

bool rain = false
for (JsonArray::iterator it=list.begin(); it!=list.end(); ++it)
{
    int weatherId = (*it)["weather"][0].id;
    if (weatherId == 500)
        rain = true;
}
```

ethelder, 2015-11-07:
Hi, 

Thank you very much for such fast and clean answer!
I'm not that good with "for" loops, and I realy need to understand my code to the very base, and so I succesfully rewrote your code.

Now I have **2 new problems**:)

1. When I try to work with the original version of this API call (3 times 3 hours) the Arduino compiler crashes. When I do it like now (which is just 3 hours from now on)- it works ok.?!
2. The loop works nice - but only once, and than it prints: `parseObject() failed` (I read the part of wiki Avoiding pitfails, but to be honest didn't understand the problem and couldn't  find the solution to this problem there - or it's too many hours in front of PC and I can't think straight:) 

My code right now is (the commented parts are there ready for the longer char):
```
#include <ArduinoJson.h>

char json[] = "{\"city\":{\"id\":7530858,\"name\":\"Poznań\",\"coord\":{\"lon\":16.901541,\"lat\":52.400631},\"country\":\"PL\",\"population\":0,\"sys\":{\"population\":0}},\"cod\":\"200\",\"message\":0.012,\"cnt\":1,\"list\":[{\"dt\":1446940800,\"main\":{\"temp\":286.55,\"temp_min\":286.549,\"temp_max\":286.55,\"pressure\":1024.19,\"sea_level\":1033.33,\"grnd_level\":1024.19,\"humidity\":95,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":5.6,\"deg\":223.502},\"rain\":{\"3h\":0.01},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2015-11-08 00:00:00\"}]}";

  
void setup() {
  Serial.begin(9600);
  while (!Serial) {
  ; // wait for serial port to connect. Needed for Yun.
  }
}

void loop() {
  parsing(json);
  delay(15000);
}

void parsing(char json1[]){
  StaticJsonBuffer<800> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json1);
  JsonArray& list = root ["list"];
  JsonArray& weather0 = list[0] ["weather"]; 
//  JsonArray& weather1 = list[1] ["weather"];
//  JsonArray& weather2 = list[2] ["weather"];
  int cnt = root ["cnt"];
  const  char* id0 = weather0[0] ["id"];
//  const  char* id1 = weather1[0] ["id"];
//  const  char* id2 = weather2[0] ["id"];
  
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
  Serial.println(id0);
//  Serial.println(id1);
//  Serial.println(id2);
  Serial.println(cnt);
  }
```

ethelder, 2015-11-07:
To be exact with the problem 1:

ok for the compiler:
```
char json[] = "{\"city\":{\"id\":7530858,\"name\":\"Poznań\",\"coord\":{\"lon\":16.901541,\"lat\":52.400631},\"country\":\"PL\",\"population\":0,\"sys\":{\"population\":0}},\"cod\":\"200\",\"message\":0.012,\"cnt\":1,\"list\":[{\"dt\":1446940800,\"main\":{\"temp\":286.55,\"temp_min\":286.549,\"temp_max\":286.55,\"pressure\":1024.19,\"sea_level\":1033.33,\"grnd_level\":1024.19,\"humidity\":95,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":5.6,\"deg\":223.502},\"rain\":{\"3h\":0.01},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2015-11-08 00:00:00\"}]}";
```

not ok for the compiler:
```
char json[] = "{\"city\":{\"id\":7530858,\"name\":\"Poznań\",\"coord\":{\"lon\":16.901541,\"lat\":52.400631},\"country\":\"PL\",\"population\":0,\"sys\":{\"population\":0}},\"cod\":\"200\",\"message\":0.012,\"cnt\":3,\"list\":[{\"dt\":1446940800,\"main\":{\"temp\":286.55,\"temp_min\":286.549,\"temp_max\":286.55,\"pressure\":1024.19,\"sea_level\":1033.33,\"grnd_level\":1024.19,\"humidity\":95,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":5.6,\"deg\":223.502},\"rain\":{\"3h\":0.01},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2015-11-08 00:00:00\"},{\"dt\":1446951600,\"main\":{\"temp\":287.87,\"temp_min\":287.869,\"temp_max\":287.87,\"pressure\":1021.76,\"sea_level\":1030.84,\"grnd_level\":1021.76,\"humidity\":95,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":9.16,\"deg\":240},\"rain\":{\"3h\":0.67},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2015-11-08 03:00:00\"},{\"dt\":1446962400,\"main\":{\"temp\":286.75,\"temp_min\":286.75,\"temp_max\":286.75,\"pressure\":1024.02,\"sea_level\":1033.03,\"grnd_level\":1024.02,\"humidity\":93,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":24},\"wind\":{\"speed\":11.42,\"deg\":281.003},\"rain\":{\"3h\":0.325},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2015-11-08 06:00:00\"}]}";
```

the compiler output:
```
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
	at java.util.regex.Pattern$BranchConn.match(Pattern.java:4568)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Branch.match(Pattern.java:4604)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
	at java.util.regex.Pattern$BranchConn.match(Pattern.java:4568)
	at java.util.regex.Pattern$CharProperty.match(Pattern.java:3777)
	at java.util.regex.Pattern$Branch.match(Pattern.java:4604)
	at java.util.regex.Pattern$GroupHead.match(Pattern.java:4658)
	at java.util.regex.Pattern$Loop.match(Pattern.java:4785)
	at java.util.regex.Pattern$GroupTail.match(Pattern.java:4717)
```

bblanchon, 2015-11-08:
When you call `parseObject()`, the input string is modified to insert null terminator and to unescape special characters.
This mean that the string isn't valid JSON after the call to `parseObject()`.
In your program, the same input string is reused, so it's only valid at the first loop iteration.

Then, for the compiler issue, I guess that the Arduino preprocessor hasn't been tested for such long lines.

Both of these problems will disappear when the JSON input will be fetched with HTTP.

bblanchon, 2015-11-08:
[Add to the FAQ](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-the-next-ones-fail)

ethelder, 2015-11-09:
Hello, once again thank you for your help and for explanation. Glad I inspired you to make your wiki more info rich. 

BTW - you say that this library works on "every arduino board" and so I would add this in your examples:
```
void setup() {
  Serial.begin(9600);
//  while (!Serial); // this line needs to be in setup for leonardo and Yun to work
// here goes the rest of your example
```
Without this `while` loop your example sketch works, but nobody can see it, since Yun doesn't restart/wait for the serial to open connection, and executes setup function right after the sketch is being uploaded to the board.

Back to my weather Yun, after some trial and error - I managed to get working this sketch (below) on Arduino Yun. Unfortunately, this is only the api call for 3 hours - 550 bytes in Json. If I would like to get any more than this (eg. 6 hours) than I need about 1000 bytes for that and when I change `char json[]` and `StaticJsonBuffer<>` to 1000 the code ... fails. This must be due to the limitation of memory which on Yun is only 2,5 K. Is there anything I could do differently to make it work? 

```
#include <ArduinoJson.h>
#include <Bridge.h>
#include <HttpClient.h>

void setup() {
  Bridge.begin();
  Serial.begin(9600);
  while (!Serial);
}

void loop() {
  char json[600];
  HttpClient client;
  int licznik = 0;
  client.get("http://api.openweathermap.org/data/2.5/forecast?id=7530858&cnt=1&appid=79485a0b05717b88a14b7b9af78faa99");
  while (client.available()) {
    char c = client.read();
    json[licznik] = c;
    Serial.print(c);
    licznik ++;
  }
  Serial.println();
 // Serial.println(json);
  parsing(json); 
  delay(60000); //minuta
}

void parsing(char json1[]){
  StaticJsonBuffer<650> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json1);
  JsonArray& list = root ["list"];
  JsonArray& weather0 = list[0] ["weather"];
//  JsonArray& weather1 = list[1] ["weather"];
//  JsonArray& weather2 = list[2] ["weather"];
  int cnt = root ["cnt"];
  const  char* id0 = weather0[0] ["id"];
//  const  char* id1 = weather1[0] ["id"];
//  const  char* id2 = weather2[0] ["id"];
  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }
  Serial.println(id0);
//  Serial.println(id1);
//  Serial.println(id2);
  Serial.println(cnt);
  }
```

bblanchon, 2015-11-12:
@ethelder You should give a try to [this other JSON library](https://github.com/squix78/json-streaming-parser/), the approach is completely different and is probably more adapted to your problem.

ethelder, 2015-11-14:
Hello, just to give you an info of how it went. At first i decided to go with the old "textfinder" library and parse this info that way but it was much too unstable so i decided to use - http get choreo on temboo and this works like a charm. It parses the information for me and passes to yun just the data that i need.

Thank you sir for your patience and for your feedback. +1 for enabling yun users to your example sketches from the first try:) THANKS

closed 2015-11-14 by ethelder