bblanchon/ArduinoJson #154: trouble with "parseObject() failed", convert const char to char

NitrofMtl, 2015-11-08:
Hi.

Just star to implant arduinoJson to my sketch. But with the parser return "parseObject() failed".

I read on the wiki that It could be cause by a read-only or pointer object. My char came from uHTTP library and is a pointer but I can't find the way to convert it to an non const var . here the code:

    void parseJSONInputs(const char json[])  {
      StaticJsonBuffer<5000> jsonBuffer;
    Serial.print(json);
    Serial.println();
      JsonObject& root = jsonBuffer.parseObject(json);

      if (!root.success()) {
        Serial.println("parseObject() failed");
        return;
      }

      for (int i = 0; i < 10 ; i++) {

        inChannelID[root["channels"][i]["canal"]].channelName = root["channels"][i]["name"];
        outChannelID[root["channels"][i]["canal"]].channelSwitch = stringToBool(root["channels"][i]["status"]);
        outChannelID[root["channels"][i]["canal"]].sp = root["channels"][i]["setPoint"];

      }
    }

bblanchon, 2015-11-09:
Well the first question is: why do you get a const string in the first place?

I don't know about uHTTP, can you provide a link to the documentation?
In particular, what function are you calling to get the JSON string.

Can you also show the code that calls `parseJSONInputs()`?

NitrofMtl, 2015-11-09:
Ok. First I must say I think I found my bug: my JSON data start with an array so I have to call parseArray instead of object... but your example don't show clearly how to parse nested array after.

But for the rest, for converting const char or char* to writable var, I found that 'strcpy()' should work I believe.

For the library uHTTP, I modify a bit the original library, here my version [uHTTP](https://github.com/NitrofMtl/uHTTP)

I'm only an hobbyist, so it a lot messy... My bad.. :P .

This is what's append when I call my JSON parser:

        else if (uHTTPserver->put_request("channels", response)) {
          parseJSONInputs(uHTTPserver->body());
        }

body() have only be modified from the original library by making it public instead of private.

     const char *uHTTP::body(){
       return __body;
     }

__body return on really big function 
    EthernetClient uHTTP::available() 
on line 169 ;
       case BODY:
        if(cr == 2 || __head.length == 0) client.flush();
        else if(cursor < uHTTP_BODY_SIZE - 1){ __body[cursor++] = c; __body[cursor] = '\0'; }
        break;

so now I think I fix the char* problem ( and the first error that I don't use the good parser... )

void parseJSONInputs(const char json[])  {
  StaticJsonBuffer<3000> jsonBuffer;
Serial.print(json);
char inchar[3000];
strcpy(inchar, json);
Serial.println();

  
  JsonArray& json_array = jsonBuffer.parseArray(inchar);

  if (!json_array.success()) {
    Serial.println("parseObject() failed");
    return;
  }

just ran some test this morning... It apprear that the JSON string is not return complete by the library. I tried to put the buffer var to fit the String but now instead of getting only the begining of the data I getting the tail...
    #define uHTTP_BUFFER_SIZE    3000
    #define uHTTP_BODY_SIZE      3000

conclusion... the main bug is not cause by arduinoJSON... lol.

But when it will work, I bwil probably need an example of parsing a nested array... my first attempt have not compiling....

Bien à vous !

Nitrof

bblanchon, 2015-11-09:
Can you please provide a link to the original uHTTP?

Using `strcpy()` is probably a mistake because it make a duplication of the body, exposing you to the risk of blowing the few kilobytes of memory available on your board.
Why don't you use `const_cast`?

BTW, the example clealy shows how to parse nested arrays:

```c++
double latitude = root["data"][0];
double longitude = root["data"][1];
```

Here `data` is a nested array of double.

If it's not enough, please check out the [wiki](https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON#subscript-operator).

NitrofMtl, 2015-11-09:
[original uHTTP](https://github.com/nomadnt/uHTTP/)

I nerver heard of const_cast... will take a look for it.

for the the nested array, sorry i'm sometime (always) a bit slow... it is maybe cause of my data structure that I'm confuse.

here a part of my json:
    {"channels":"[{\"name\":\"name0\",\"canal\":\"0\",\"status\":true,\"temperature\":21.4,\"setPoint\":5,\"permission\":false,\"percentOut\":0,

so I've got 10 channels.. I just have to do:
    int setpoint = root[channels][0][setpoint];

I'm right ?

thank.

bblanchon, 2015-11-09:
After a quick glance, it looks like it's OK to do:

```c++
JsonObject& root = jsonBuffer.parseObject(static_cast<char*>(uHTTPserver->body()));
```

`static_cast` just removes the constness of it's argument. In other word, it turns the security off.

If you need to iterate through every channel, you'd better use iterators:

```c++
JsonArray& channels = root["channels"];
for (JsonArrar::iterator it=channels.begin(); it!=channels.end(); ++it) {
   JsonObject& channel = *it;
   int setpoint = channel["setpoint"];
   // ...
}
```

I'm curious, what kind of thermostat are you driving?

domonetic, 2015-11-09:
why noto sent as nestedarray the 10 channels

¿y porque no enviar los datos de los 10 canales como un array?
Name
TempChannels[0,1,2,3,4,5,6,7,8,9]
status
setPoint
permission
percentOut

NitrofMtl, 2015-11-09:
@bblanchon ok... will try it for sure. a soon as I fix my 'body()' problem. hummm.

To fill up your curiosity, the arduino will became the thermostat, it have 10 temp probe on analog input, and have 10 ssr output for heater, control trough digital pin with a duty cycle of 10 second. 

@domonetic it was easier for me to work with nested channels because 1- internal data management true my struct and the parse it to my web page using angular...

NitrofMtl, 2015-11-09:
JsonObject& root = jsonBuffer.parseObject(static_cast<char*>(uHTTPserver->body()));

error compiling

    demo_webserv9debug.ino.ino: In function 'void parseJSONInputs()':
demo_webserv9debug.ino:576: error: invalid static_cast from type 'const char*' to type 'char*'

bblanchon, 2015-11-17:
@NitrofMtl I'm really sorry I forgot to answer.

I wrote a mistake in the sample code above, here is the correct answer:

```c++
JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(uHTTPserver->body()));
```

In C++, only `const_cast` is allowed to removes the constness of a variable.

NitrofMtl, 2015-11-17:
Thank. 

One other question. Is it possible to make bigger max oblect sixe. It appear that I cant make string bigger thant 6:

 *   "channels": {
    "setPoint": 5,
    "name": "foobar",
    "canal": "4"
  }
}parseObject() failed //if it is foobar1 for example

bblanchon, 2015-11-17:
I'm sorry, I don't understand. Can you rephrase?

NitrofMtl, 2015-11-17:
on      JsonObject& root = jsonBuffer.parseObject(json); 
if for example my object 'Name' is bigger than 6 char, the parse obj fail. it work woth  "name": "foobar" and not with  "name": "foobarr"

bblanchon, 2015-11-19:
What is the type of `json`?
If it's a `String` or a `const char*`, the `JsonBuffer` have to make a copy of the string, and that would explain you problem.

NitrofMtl, 2015-11-19:
void parseJSONInputs(const char json[])  {
      StaticJsonBuffer<200> jsonBuffer;
      JsonObject& root = jsonBuffer.parseObject(json);
    
      if (!root.success()) {
        Serial.println("parseObject() failed");
        return;
      }
    
      int  chanId = root["channels"]["canal"];
      inChannelID[chanId].channelName = toString(root["channels"]["name"]);
      outChannelID[chanId].channelSwitch = stringToBool(root["channels"]["status"]);
      outChannelID[chanId].sp = atoi(root["channels"]["setPoint"]);
    root.prettyPrintTo(Serial);
    }

if I understand what you said before, I can do:

     JsonObject& root = jsonBuffer.parseObject(const_cast<char*>json);

NitrofMtl, 2015-11-19:
Tested and working with a small corection

    parseObject(const_cast<char*>(json));

Thank a lot for your time !

closed 2015-11-23 by bblanchon

bebeno, 2017-06-06:
Hi guys. 

I am new in this problematic. I would like to ask you about help with json string and how parse it. 
I need get data from ESP8266 flashed with espeasy firmware.  
This is my code ( just json and parse part) 

`WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // We now create a URI for the request
  String url = "/json?";
  Serial.print("Requesting URL: ");
  Serial.println(url);
  // This will send the request to the server
  client.print("GET " + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  delay(500);
  char c[1024];
  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    c[0] = client.read();

    //Serial.print(c);

    Serial.print(c);
  }
  StaticJsonBuffer<250> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(c);

  if (!root.success())
{
  Serial.println("parseObject() failed");
  return;
}
  //const char* sensors = root["Sensors"];
  float temp = root ["Sensors"]["Temp-voda"];
  Serial.println();
 // Serial.print(sensors);
  Serial.print(temp);

  Serial.println();
  Serial.println("closing connection");
}

My problem is that everythime I get parse Object faild. 

This is json string from 
`{"System":{
"Build": 145,
"Unit": 2,
"Uptime": 139,
"Free RAM": 26472
},
"Sensors":[
{
"TaskName": "Voda",
"Temp-voda": 20.44
},
{
"TaskName": "Wifi",
"wifi": -66.00
},
{
"TaskName": "Spiral",
"spiral": 1.00
}
]}`

Any idea what can by wrong )?
Thanks

bblanchon, 2017-06-08:
Hello @bebeno,

I think the `JsonBuffer` is too small.

Please read:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)

Regards,
Benoit

beicnet, 2017-11-28:
I don't know what I doing wrong, but I got "parse object failed error" too, and yes I read the up mentioned FAQ...

```
#include "ESP8266WiFi.h"
#include "ESP8266HTTPClient.h"
#include "ArduinoJson.h"

const char* ssid = "Test";
const char* password = "1234567890";

String JsonData = "";

void setup ()
{
    Serial.begin(9600);
    WiFi.begin(ssid, password);
    
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(1000);
        Serial.print(".");
    }
    Serial.println("Connected!");
}

void loop()
{
    
    if (WiFi.status() == WL_CONNECTED) {
        
        HTTPClient http;
        
        http.begin("http://jsonplaceholder.typicode.com/users/1");
        int httpCode = http.GET();
        
        if (httpCode > 0) {
            
            JsonData = http.getString();
            Serial.println(JsonData);
            
            StaticJsonBuffer<800> jsonBuffer;
            JsonObject& root = jsonBuffer.parseObject(JsonData);
            
            if (!root.success()) {
                Serial.println("parseObject() failed");
                return;
            }
            
            const char* stname = root["name"];
            Serial.println(stname);
        }
        
        http.end();
        
    }
    delay(10000);
}
```
Thank you for your help! ;)

bblanchon, 2017-11-28:
Hi @beicnet,

`typicode.com` recently switched to HTTPS only, so the example doesn't work anymore.

You can grab the updated version of the example from the master branch:
https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino

The new example will be shipped with the next version of ArduinoJson.
Sorry for the inconvenience.

Regards
Benoit

beicnet, 2017-11-28:
Wow @bblanchon Thank you for the epic fast response! ;)

Sorry, but I don't understand you! :( And why is that concerning my example sketch?

Kind regards,
Viktor

bblanchon, 2017-11-28:
The problem is that `httpCode` is a `301` and there is no JSON in the response.
The code you mentioned was probably copied from the old `JsonHttpClient.ino`.
Have a look a the new version to fix your code.

Sorry for the confusion, it's been a long day 😫

beicnet, 2017-11-28:
> Sorry for the confusion, it's been a long day :tired_face:

Not a problem, just relax! ;)

Then, why I'm getting the correct reply by the GET?

`JsonData = http.getString();`

and the output is a nice and clean string:
```
{
  "id": 1,
  "name": "Leanne Graham",
  "username": "Bret",
  "email": "Sincere@april.biz",
  "address": {
    "street": "Kulas Light",
    "suite": "Apt. 556",
    "city": "Gwenborough",
    "zipcode": "92998-3874",
    "geo": {
      "lat": "-37.3159",
      "lng": "81.1496"
    }
  },
  "phone": "1-770-736-8031 x56442",
  "website": "hildegard.org",
  "company": {
    "name": "Romaguera-Crona",
    "catchPhrase": "Multi-layered client-server neural-net",
    "bs": "harness real-time e-markets"
  }
}
```
and I wish to parse that string output...

Maybe I'm too n00b to catch you! :(

bblanchon, 2017-11-28:
Whaaaaat! I can't believe they fixed this!
OK cool.

I happened to have an ESP8266 connected right now:
[image]

As you can see it works here.

I don't know what's happening, can you check the library and the core version?
Also, can you try with a `DynamicJsonBuffer`?

beicnet, 2017-11-28:
I told you!!! :D It have been working a long time ago!

I'm using all old versions, I think, but let me check it for you!

ESP8266 Arduino core version: 2.3.0
ArduinoJson version: 5.6.7

So, my result now with your proposal with the: `DynamicJsonBuffer` is working fine...

But, please tell me how my sketch working on your side with `StaticJsonBuffer` and you rather got correct result and not parse object error?!

Can you please explain?