bblanchon/ArduinoJson #155: Exception (29) thrown using Adafruit HUZZAH ESP8266 board

CharlesRay, 2015-11-09:
Running your JasonParserExample I get an exception (29) with the following dump.  I did not modify the code except for increasing the baud rate to 115200.  Code looks great, I look forward to using it.

Exception (29):
epc1=0x4020430f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffe9ad0 end: 3ffe9e00 offset: 01a0

>>>stack>>>
3ffe9c70:  3ffe9e58 00000000 3ffe9e24 3ffe9e2c  
3ffe9c80:  3ffe9d77 3ffe9e58 3ffe9d80 4020275c  
3ffe9c90:  3ffe9e58 3ffe9e58 3ffe9ca4 40202184  
3ffe9ca0:  3ffe8898 3ffe9ca0 3ffe9cac 3ffe9cbc  
3ffe9cb0:  3ffe9d70 00000002 3ffe9d77 3ffe9cec  
3ffe9cc0:  3ffe9d7b 00000001 3ffe9d80 3ffe9ca0  
3ffe9cd0:  3ffe9cd4 3ffe9ce0 00000001 3ffe9d90  
3ffe9ce0:  00000000 00000001 3ffe9d9a 00000000  
3ffe9cf0:  3ffe9d8b 00000005 3ffe9ccc 00000000  
3ffe9d00:  00000000 00000000 00000000 00000000  
3ffe9d10:  00000000 00000000 00000000 00000000  
3ffe9d20:  00000000 00000000 00000000 00000000  
3ffe9d30:  00000000 00000000 00000000 00000000  
3ffe9d40:  00000000 00000000 00000000 00000000  
3ffe9d50:  00000000 00000000 00000000 00000000  
3ffe9d60:  00000000 00000000 00000000 00000058  
3ffe9d70:  736e6573 6700726f 74007370 00656d69  
3ffe9d80:  31353331 31343238 64003032 00617461  
3ffe9d90:  372e3834 38303635 2e320030 30323033  
3ffe9da0:  2c003833 30332e32 38333032 00007d5d  
3ffe9db0:  00000000 00000000 00000000 00000000  
3ffe9dc0:  00000000 00000000 00000000 00000000  
3ffe9dd0:  00000000 00000000 00000000 3ffe9e2c  
3ffe9de0:  3fffdc20 00000000 3ffe9e24 4020183f  
3ffe9df0:  00000000 00000000 3ffe8de0 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

bbx10, 2015-11-09:
The solution is to use the ESP8266 staging version https://github.com/esp8266/arduino#staging-version- which has a fix for this problem.

bblanchon, 2015-11-09:
[Now in the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#esp8266)

closed 2015-11-09 by bblanchon