bblanchon/ArduinoJson #157: error: 'namespaceArduinoJson' does not name a type

bblanchon, 2015-11-09:
When compiling the examples on Arduino 1.6.6 on Windows, there is the following error:

```
JsonGeneratorExample:9: error: 'namespaceArduinoJson' does not name a type
```

bblanchon, 2015-11-09:
It's a [known bug](JsonGeneratorExample:9: error: 'namespaceArduinoJson' does not name a type) in Arduino 1.6.6.

The solution is to roll back to 1.6.5 or use the hourly build.

domonetic, 2015-12-01:
any solution?

bblanchon, 2015-12-01:
As I said, for now you have to go back to 1.6.5 or download the [hourly build](http://www.arduino.cc/en/Main/Software#hourly).

domonetic, 2015-12-01:
yes, in hourly build run fine.

bblanchon, 2015-12-22:
Fixed in Arduino 1.6.7

closed 2015-12-22 by bblanchon

gsalvati, 2016-02-01:
Still the same problem for me:

SRC:
JsonObject& deviceObj = jsonBuffer.createObject();
deviceObj["type"] = "switch";

Error:
error: 'deviceObj' does not name a type
 deviceObj["type"] = "switch";
 ^
exit status 1
'deviceObj' does not name a type

bblanchon, 2016-02-01:
@gsalvati: I don't think it's related, can you please open a new issue and paste a bit more code?

gsalvati, 2016-02-02:
@bblanchon orry, I found the problem, deleted the post. My bad

ksmiller99, 2018-06-26:
@gsalvati - what was the error and how was it resolved?