bblanchon/ArduinoJson #158: Compiler warnings: always_inline function might not be inlinable

andig, 2015-11-12:
At https://github.com/esp8266/Arduino/issues/978 I've noticed compiler warnings when using the xtensa toolchain:

	C:\Users\Documents\Arduino\libraries\ArduinoJson\src\JsonVariant.cpp:38:18: warning: always_inline function might not be inlinable [-Wattributes]
	 FORCE_INLINE int parse<int>(const char *s) {
			  ^
	C:\Users\Documents\Arduino\libraries\ArduinoJson\src\JsonVariant.cpp:33:19: warning: always_inline function might not be inlinable [-Wattributes]
	 FORCE_INLINE long parse<long>(const char *s) {
			   ^
	C:\Users\Documents\Arduino\libraries\ArduinoJson\src\JsonVariant.cpp:28:21: warning: always_inline function might not be inlinable [-Wattributes]
	 FORCE_INLINE double parse<double>(const char *s) {
			     ^
	C:\Users\Documents\Arduino\libraries\ArduinoJson\src\JsonVariant.cpp:23:20: warning: always_inline function might not be inlinable [-Wattributes]
	 FORCE_INLINE float parse<float>(const char *s) {

bblanchon, 2015-11-12:
This has already been fixed in version 5.0.6

andig, 2015-11-12:
Thank you. 5.0.5 is currently the highest version visible in the Arduino IDE.

andig, 2015-11-12:
There is a follow-on error actually:

    C:\Users\Documents\Arduino\libraries\ArduinoJson\test\ArduinoString_Tests.cpp:7:25: fatal error: gtest/gtest.h: No such file or directory
     #include <gtest/gtest.h>

Removing the test folder takes care of that.

bblanchon, 2015-11-12:
It should be 5.0.6.
Can you try to flush the cache?
[Intructions](https://forum.arduino.cc/index.php?topic=350901.msg2419474#msg2419474)

andig, 2015-11-12:
That fixed it- confusing as the startup-update doesn't seem to do the trick. Thank you!

closed 2015-11-12 by andig

bblanchon, 2015-11-12:
Thank you very much for the feedback.