bblanchon/ArduinoJson #159: jsonp callback function

albandes, 2015-11-13:
Hi.

It is possible to add the callback so they can be made croos-domain requests?

Example:

CallBackString([{"device":"Temperatura","status":"OK","value":0.34}])

Tks

bblanchon, 2015-11-13:
Hi,
Yes it's possible with a bit of imagination :wink: 

```c++
StaticJsonBuffer<200> jsonBuffer;

JsonObject& root = jsonBuffer.createJsonObject();
root["device"] = "Temperatura";
root["status"] = "OK";
root["value"] = 0.34;

Serial.print("CallBackString([");
root.printTo(Serial);
Serial.println("])");
```

Regards.

albandes, 2015-11-13:
Hi.

I have a webserver in Arduino and return jsonp. I have a lot of problem (memory fragment) because I use string object. Now i use your bib and char. But concat char variable is a problem for me.

Tks

bblanchon, 2015-11-13:
In the example above, there is neither memory fragmentation (because there is no dynamic allocation), nor `String`, nor `strcat()`.

What are you talking about?

albandes, 2015-11-13:
Hello

Now I get it.

Thank You

Regards

closed 2015-11-17 by bblanchon