bblanchon/ArduinoJson #162: NULL return 0 but null value exist in json

SnouF, 2015-11-19:
Hello

By example

  root["null"] = NULL
return
  {"null":0}

If it could return
  {"null":null}
would better right?

bblanchon, 2015-11-20:
Hi Jonas,

UnfortunateIy, this is not possible as `NULL` is an alias for `0`.
Look at [the definition of `NULL`](http://fr.cppreference.com/w/cpp/types/NULL):

```c++
#define NULL 0
```

However ArduinoJson will behave as you expect if you give a null string pointer:

```c++
root["null"] = static_cast<const char*>(NULL);
```

domonetic, 2015-11-20:
Can use NIL u other word

bblanchon, 2015-11-20:
@domonetic I've never heard of a `NIL`in C++...

domonetic, 2015-11-20:
I misinterpreted the question

SnouF, 2015-11-20:
Tanks bblanchon

Sorry for this "impossible feature request" :)

closed 2015-11-20 by SnouF

bblanchon, 2015-11-20:
That's OK Jonas :wink: 
Thank you for using ArduinoJson.