bblanchon/ArduinoJson #163: Trouble with Doubles

fcgdam, 2015-11-19:
Not sure if an issue, my fault or just a bug... ;) anyway:

Suppose that I have the following json { 'val':15}  where the val value can be 15 without comma without any decimal places or 15.1, for example.
With this case casting either to int or to double fails depending on the number format returned.
For example:
double temp = temp["temp"].as<double>(); 
will return 0 if the value is 15, and return 15.1 if the value is 15.1
Of course if the cast is to int, then it fails the other way around...
Any ideas in how to solve this issue? I can't control the returning Json value format.

bblanchon, 2015-11-20:
This really looks like a bug.
Can you please copy/paste a short code example?

fcgdam, 2015-11-20:
Sure: First just some quick notes to explain what I'm doing:
 The code is for decoding the JSON response from openweathermap services.
 The temperature, for example, when is an exact number it returns 15 instead of 15.0 for example.
 But the temperature can also be 15.4 or 15.21 for example. So the number format returned varies. 
 You can make a demo call here:
 
 http://api.openweathermap.org/data/2.5/weather?q=London,uk&units=metric&appid=2de143494c0b295cca9337e1e96b00e0

There are other properties that have the same behaviour like wind speed, pressure...

I'm note sure also what version of the Library I'm using but I suppose it is R4 that is used in Sming framework used to build firmware for the ESP8266 chip: https://github.com/SmingHub/Sming

So my code is this one: http://pastebin.com/TZzNGyqZ

The code has a workaround the getValue function, that tries to decode twice the value of the result is zero. With that function things work as expected, but this issue will caught many people off guard like I was :)

bblanchon, 2015-11-20:
I'm not able to reproduce the bug on a PC.
See [this test](https://github.com/bblanchon/ArduinoJson/blob/77f1d79225acbf14247b4542983ed07059db745c/test/Issue163.cpp) is passing:

    DynamicJsonBuffer jsonBuffer;
    char json[] = "{'val':14}";
    JsonObject &root = jsonBuffer.parseObject(json);
    ASSERT_EQ(14.0, root["val"].as<double>());

It must be somehow related to ESP8266.
Can you try with `float` instead of `double`?
Can you try to call `strtod()` from your code and check if it works as expected?

fcgdam, 2015-11-20:
I'll have to try it later today.
I also have some Arduinos, and I'll check if with that platform if the issue is also there, but I also suspect that is also ESP8266 related, compiler issues? Who knows.

bblanchon, 2015-11-20:
It's surely not a compiler issue but it could definitely come from the [Arduino core for ESP8266](https://github.com/esp8266/Arduino).
They wrote their [own `strtod()`](https://github.com/esp8266/Arduino/blob/4cf72e7ef4fcabde230de44ba1d1ab5d57cdeee0/cores/esp8266/libc_replacements.c#L225) and it has [already been a source for trouble](https://github.com/bblanchon/ArduinoJson/issues/101).

A simple test like:

    double val = strtod("14",0);

should tell if it works or not.

By the way, are you using the "stable" or the "staging" version of the ESP8266 core?

fcgdam, 2015-11-20:
Hi: 
I'm not using the Arduino core for the ESP8266 but an alternative framework, also for the ESP8266, called Sming. I'm going to check what kind of implementation they have on strtod, if a known one or their own implementation.

fcgdam, 2015-11-20:
Humm... Not the strtod function to be the culprit.  It works fine. With float it also has the same behaviour. These tests with Sming framework and the ESP8266:
<PRE>
    Serial.println("Teste");
    double val = strtod("14",0);
    Serial.println("Valor:" + String(val));
    Serial.println("Valor:" + String(val+1));
 
    DynamicJsonBuffer jsonBuffer;
    char json[] = "{'val':14}";
    JsonObject &root = jsonBuffer.parseObject(json);
    Serial.println("Json: " + String(root["val"].as<double>()));
    
    Serial.println("Teste2");
    float val1 = strtod("14",0);
    
    Serial.println("Valorf:" + String(val1));
    Serial.println("Valorf:" + String(val1+1));
    
    Serial.println("Json: " + String(root["val"].as<float>()));
</PRE>
This gives:
<PRE>
Teste
Valor:14.00
Valor:15.00
Json: 0.00
Teste1
Valorf:14.00
Valorf:15.00
Json: 0.00
</PRE>
So not the function strtod.
Let me see with the Arduino framework.

bblanchon, 2015-11-20:
On an Arduino Duelmilanove:
```
Teste
Valor:14.00
Valor:15.00
Json: 14.00
Teste2
Valorf:14.00
Valorf:15.00
Json: 14.00
```

bblanchon, 2015-11-20:
OK, I found the issue: [Sming includes version 4.2 of ArduinoJson](https://github.com/SmingHub/Sming/tree/a80fdd7166afafa34ad78395d4f1a77631fbbc90/Sming/Services/ArduinoJson)

The behavior you're seeing was the one implemented in ArduinoJson 4.x.
Changing this was one of the major motivations behind version 5.x.

Upgrading to 5.x is the best thing you could do: simpler API, better performance and smaller code size.

fcgdam, 2015-11-20:
Same results for me on the UNO, just like your Arduino.

Edit: Haah, OK, I think the develop branch of Sming does use R5. Not sure exactly what sub version:

https://github.com/SmingHub/Sming/commit/98b662c397e8de8e677c31f9eca61a464e26a037

Let me see if I can compile under the latest version to see if the issue is gone.

fcgdam, 2015-11-20:
Ok. The Sming develop branch uses ArduinoJson R5 but it has some serious problems with the flashing tools. Anyway, I was able to build an image and flash it on the ESP8266 using alteranative tool than the default, and it works.
So using R5 of ArduinoJson library works as expected :) and the above code produces the correct output:
<PRE>
Teste
Valor:14.00
Valor:15.00
Json: 14.00
Teste
Valorf:14.00
Valorf:15.00
Json: 14.00
</PRE>

So the issue is explained, I have two work arounds: Use R5 or use my getValue function.
Thanks for your support.

To conclude and

closed 2015-11-20 by fcgdam