bblanchon/ArduinoJson #165: Adding new object to array dynamically

UICJohn, 2015-11-22:
HI, I am trying to add new json object to json array dynamically. To making my arduino easy to add new sensor, I need to dynamically create new json object:

 while(counter<sensor_cont) 
  {
    SensorObj sensor = nextSensorInf(&index); \\This will retrieve next sensor information from EEPROM
    JsonObject& obj = jsonBuffer.createObject();
    obj["label"] =  sensor.label;
    obj["reading"] = sensor.type == 0? analogRead(sensor.pin):digitalRead(sensor.pin);
    data.add(obj);
    counter++;  
  }
However, this does not work at all.  How can I add new json object correctly?  THX : )

bblanchon, 2015-11-22:
Hi,

The approach is correct.
However, you may have a problem with the following line:

    obj["label"] = sensor.label;

If `label` is of type `char*`, the object will store a reference to the string.
See [Avoiding pitfall: Do not assume that strings are copied](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#6-do-not-assume-that-strings-are-copied).
A good workaround would be:

    obj["label"] = jsonBuffer.strdup(sensor.label);

Lastly, you can shorten the code by using `createNestedObject()`:

```c++
for (count = 0; counter < sensor_count; counter++) 
{
  SensorObj sensor = nextSensorInf(&index);
  JsonObject &obj = data.createNestedObject();
  obj["label"] = jsonBuffer.strdup(sensor.label);
  obj["reading"] = sensor.type == 0 ? analogRead(sensor.pin) : digitalRead(sensor.pin);
}
```

UICJohn, 2015-11-23:
Yeah, It works! Thank you very much! Cheers!

closed 2015-11-23 by UICJohn

pkourany, 2016-02-26:
@bblanchon, I want to add N number of nested array (each with 3 items) from a file.  The version of ArduinoJson I have (ported to Particle platform) does not include strdup (v4?).   I tried the multi-dimensional array until I realized that the values would not perpetuate when the function the code below was in, finished.  Is there another way to achieve this?
```
char fld[ADC_DIM+1][20];

	while (notEOF) {
		JsonArray &dataSet = values.createNestedArray();
		for (uint8_t i = 0; i < num_channels+1; i++) {
			if (readField(&logfile, fld[i], sizeof(fld[i]), ",") == 255) {	// done if Error or at EOF
				notEOF = false;
				break;
			}
			char* dupl = object.strdup(fld[i]);  // Doesn't compile
			dataSet.add(dupl);
		}
	}
```
Thanks!

bblanchon, 2016-02-26:
@pkourany From what I can see here, you don't need `strdup()` since each string is stored in a different location inside the `fld` array.

pkourany, 2016-02-26:
@bblanchon, problem is, the array sizing is only good enough for one 3-member array and the number of those on the SD is unknown.

bblanchon, 2016-02-26:
@pkourany You can make it bigger (and probably move it to the heap).
With don't you upgrade to ArduinoJson 5.1 anyway? It's better in every possible way :wink:

pkourany, 2016-02-26:
@bblanchon, I could make it bigger (eg. max possible size) and on the heap.   As for upgrading, the [Particle platform](https://www.particle.io/) (STM32) presently does not support libraries or sub-directories in the make files (flat application folder).  I'm not sure if there will be an issue with .hpp and .ipp files as well.  To make a long story short, without modifying the make files, the ArduinoJson library files will need lots of mods to work.  The port I am using was done by someone else.  I will look into porting nonetheless.