bblanchon/ArduinoJson #166: printTo(char * but, int from, int to)

sticilface, 2015-11-23:
Revisiting this issue. Feature request to allow printing of segments of json string to buffer, on esp8266 using arduinoIDE.

Why would this be useful.  It would allow you wrap the printTo in a loop that would copy chunks of the jsonstring to a buffer to then be sent out using WiFiClient write, or web server sendcontent. At the moment if printTo is used 
```
        WiFiClient c = HTTP.client();
        root.printTo(c);
```
it results in a packet for each char, i've checked using wireshark, literally 1000's of packets for a long string.  this is very inefficient and slow.  
The workaround, is to copy it all to a buffer, but then you are now holding two copies in memory, and this is also not very suitable.  If you can copy chunks to a buffer of known size, say 1440B then you can loop through, send optimum packet sizes, and keep the memory overhead down?

sticilface, 2015-11-23:
I just realised my autocorrect changed buf to but in the header. Oops. At least it is not butt!!

bblanchon, 2015-11-23:
What you need is a proxy between ArduinoJson and WifiClient that will do the buffering:

```c++
WiFiClient client = HTTP.client();
BufferedPrint<512> proxy(client);
root.printTo(proxy);
proxy.flush();
```

Here is a possible implementation (not tested):

```c++
template<size_t CAPACITY>
class BufferedPrint : public Print
{
public:
    BufferedPrint(Print & destination) : _destination(destination), _size(0)
    {
    }

    virtual size_t write(uint8_t c)
    {
        _buffer[_size++] = c;

        if (_size + 1 == CAPACITY)
        {
            flush();
        }
    }

    void flush()
    {
        buffer[_size] = '\0';
        _destination.print(_buffer);
        _size = 0;
    }

private:
    Print & _destination;
    size_t _size;
    char _buffer[CAPACITY];
};
```

sticilface, 2015-11-23:
That looks very promising.  Thank you!  will try later

sticilface, 2015-11-24:
This works very well.  Thank you.  here is what I adapted it to. 
```
template<size_t CAPACITY>
class BufferedPrint : public Print
{
public:
    BufferedPrint(ESP8266WebServer & HTTP) : _HTTP(HTTP), _size(0)
    {
    }

    virtual size_t write(uint8_t c)
    {
        _buffer[_size++] = c;

        if (_size + 1 == CAPACITY)
        {
            flush();
        }
    }

    void flush()
    {
        _buffer[_size] = '\0';
        _HTTP.sendContent( String(_buffer)); 
        _size = 0;
    }

private:
    ESP8266WebServer & _HTTP; 
    size_t _size;
    char _buffer[CAPACITY];
};
```
then to send 
```
        size_t jsonlength = root.measureLength();
        _HTTP.setContentLength(jsonlength); 
        _HTTP.send(200, "text/json" ); 
        BufferedPrint<1440> proxy(_HTTP);
        root.printTo(proxy);
        proxy.flush();
```

bblanchon, 2015-11-24:
Cool!

closed 2015-11-24 by bblanchon

sticilface, 2015-11-26:
Sorry to post again, but I thought i'd share some more optimisations.  I tried to get it to work initially with WiFiClient, but failed as the compiler borked and gave some strange errors i could not interpret..  A few days later I thought I'd try a bit of casting magic, and it works.  The new code will take either the ESP8266WebServer, or WiFiClient and send out buffered packets...  

```
template<size_t CAPACITY>
class BufferedPrint : public Print
{
public:
	BufferedPrint(ESP8266WebServer & HTTP) : _size(0)
	{
		_client = HTTP.client();
	}
	
	BufferedPrint(WiFiClient & client) : _client(client), _size(0)
	{
	}

	~BufferedPrint() {
		_client.stop(); 
	}

	virtual size_t write(uint8_t c)
	{
		_buffer[_size++] = c;

		if (_size + 1 == CAPACITY)
		{
			flush();
		}
	}

	void flush()
	{
		 _client.write( (const char *)_buffer, _size);
		 _size = 0;
	}

private:
	WiFiClient _client;
	size_t _size;
	char _buffer[CAPACITY];
};

```

mayankraichura, 2016-06-09:
I've been using PString by Mikal Hart.

    char json_out_buffer[500];
    PString jsonString(json_out_buffer, sizeof(json_out_buffer));
    obj.printTo(jsonString);
    server.send(200, c_json, json_out_buffer);

mdvorak, 2016-12-12:
I'm gonna add my version, since its still needed when used on ESP. My comments to previous example:
* `if (_size + 1 == CAPACITY)` - there is no need to add one, it will always leave 1 byte of buffer unused.
* Calling `client.stop` in destructor might have side effect - it might be too late or too early. Don't do that, leave that to the caller. I would not even call flush, because you should already have called it before calling stop. Your code still forces caller to call flush, but magically closes the client. Thats not good.
* Class name `BufferedPrint` does not reflect what this does - a bridge between WiFiClient and Print interface (btw dunno why WiFiClient does not implement it, there is no reason for that).

I've written this independently on this issue, but given comments above, I'm sharing it here, despite being almost identical.
Destructor for debug builds is there for fail-fast paradigm. Forces you to flush it (or close it) explicitly.

```
#pragma once
/// This is intended for ESP8266 Arduino stack

#include <WiFiClient.h>
#include <Print.h>

template<size_t BUFFER_SIZE = 32>
class WiFiClientPrint : public Print
{
  public:
    WiFiClientPrint(WiFiClient client)
      : _client(client),
        _length(0)
    {
    }

    ~WiFiClientPrint()
    {
#ifdef DEBUG_ESP_PORT
      // Note: This is manual expansion of assertion macro
      if (_length != 0) {
        DEBUG_ESP_PORT.printf("\nassertion failed at " __FILE__ ":%d: " "_length == 0" "\n", __LINE__);
        // Note: abort() causes stack dump and restart of the ESP
        abort();
      }
#endif
    }

    virtual size_t write(uint8_t c) override
    {
      _buffer[_length++] = c;
      if (_length == BUFFER_SIZE) {
        flush();
      }
    }

    void flush()
    {
      if (_length != 0) {
        _client.write((const uint8_t*)_buffer, _length);
        _length = 0;
      }
    }

    void stop()
    {
      flush();
      _client.stop();
    }

  private:
    WiFiClient _client;
    uint8_t _buffer[BUFFER_SIZE];
    size_t _length;
};
```

Usage:
```
// ESP8266WebServer _server;
// JsonObject json;
_server.setContentLength(json.measureLength());
_server.send(200, "application/json", "");

WiFiClientPrint<> p(_server.client());
json.printTo(p);
p.stop(); // Calls p.flush() and WifiClient.stop()
```

I’m suggesting linking this issue to Bag of tricks or Compatibility issues Wiki pages. It took me a while to find out best way to send json on ESP, including need for own buffer. Or include it in the sources until WiFiClient will implement Print on its own.

bblanchon, 2016-12-14:
Hi Michal,

The `BufferedPrint` in the Bag Of Trick is more generic, as it works with any `Print` implementation.
So I think we should keep it.

However, your `WiFiClientPrint` has a real value, so I think we should add it to the Bag Of Tricks, just after the `BufferedPrint`

Can you edit the wiki?

mdvorak, 2016-12-18:
Added to wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#render-to-esp8266-wificlient

I've left the BufferedPrint intact, feel free to change it/combine it etc.

bblanchon, 2016-12-18:
Thank you very much Michal 👍