bblanchon/ArduinoJson #167: Parsing Arrays Fails with Github API

tommasoltrz, 2015-11-23:
Hi !
I'm trying to decode github API using the library, but I'm having some trouble.
I've tried several solution, like using parseObject instead of parseArray, but it doesn't work either, i need it to be an Array so I can get nested object and stuff.

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n"+
               "Connection: close\r\n\r\n");
    Serial.println("request sent");
    while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
    }
    String line = client.readStringUntil('\n');

    JsonArray& root = jsonBuffer.parseArray(line);  

    if (!root.success()) {
    Serial.println("parseObject() failed");
      }
     long id = root[0]["id"];
     Serial.println(id);

Cheers

bblanchon, 2015-11-23:
Hi,

I'll be very glad to help you, but I would need more information:

1. What contains `line` before the call to `parseArray()` ?
2. What is the type of `jsonBuffer`  ?

Regards,

tommasoltrz, 2015-11-23:
Yes, sure.
line contains the whole string returned by github,
let's say, something like this:
https://api.github.com/orgs/casajasmina/events?page=1&per_page=1

JsonBuffer is Static with 20000 of space,
(I tried to give it the maximum size, but perhaps is not necessary)

bblanchon, 2015-11-23:
1. Please print the actual string from your code as it may be wrong
2. 20KB is too big for the ESP8266 stack, you should switch to `DynamicJsonBuffer`

tommasoltrz, 2015-11-23:
1. The line string actually contains the entire result I get from the github api. 
I tested it by printing it on the serial port, and the content is exactly the same of what you see [here](https://api.github.com/orgs/casajasmina/events?page=1&per_page=2). When calling the API from the browser

2. I'm actually using an Arduino zero board with a wifi shield 101, so 20kb should be ok as it has 32kb of SRAM.

bblanchon, 2015-11-23:
Are you sure?
Because the JSON is full of line breaks.

To me, the following line:

    String line = client.readStringUntil('\n');

will read only one line, not the whole JSON.
So I bet the actual content of `line` is:

    [

closed 2015-12-01 by bblanchon