bblanchon/ArduinoJson #168: Fail to parse using parseArray

frippe75, 2015-11-24:
Probably more of a question rather than an issue... Not sure!
Not sure how to parse this using ParseArray...
Outputs nothing. I want to find out that the latest firmware of the esp8266 is "esp8266-0.02-1.bin"

JSON:
{
    "esp8266": [
        {
            "version": "latest",
            "file": "esp8266-0.02-1.bin"
        },
        {
            "version": "0.02-1",
            "file": "esp8266-0.02-1.bin"
        },
        {
            "version": "0.01-1",
            "file": "esp8266-0.01-1.bin"
        }
    ],
    "ATmega": [
        {
            "version": "latest",
            "file": "atmega-0.01-1.hex"
        },
        {
            "version": "0.01-1",
            "file": "atmega-0.01-1.hex"
        }
    ]
}

Arduino/ESP8266 Code:
     StaticJsonBuffer<400> jsonBuffer;
     char json[300];
          JsonArray& root = jsonBuffer.parseArray(json,2);
if (root.success())
     {
        //for (JsonObject::iterator it=root.begin(); it!=root.end(); ++it)
        for (JsonArray::iterator it=root.begin(); it!=root.end(); ++it)
        {
          //USE_SERIAL.println(it.);
          /*
          if (it == "Esp8266") {
            
          }
          */
          /*
          USE_SERIAL.println("KEY");
          USE_SERIAL.println(it->key);
          USE_SERIAL.println("VALUE");
          USE_SERIAL.println(it->value.asString());
          */
        }
        
     } else {
        // TODO: Handle error
    }

bblanchon, 2015-11-24:
```c++
for (JsonObject::iterator it=root.begin(); it!=root.end(); ++it)
{
    if (!strcmp(it->key, "esp8266"))
    {
        const char* version = it->value["version"];
        // ...
    }
}
```

frippe75, 2015-11-24:
Thanks for the quick reply!
I did a small sketch to test it out. 

It iterates both elements (esp8266,atmega) finds the esp8266 and matches the version value keypair. version is empty though.

**Outputs:**
Iterate
version=

Iterate

```
  StaticJsonBuffer<500> jsonBuffer;

  char json[] =
      "{\"esp8266\":[{\"version\":\"latest\",\"file\":\"esp8266-0.02-1.bin\"},{\"version\":\"0.02-1\",\"file\":\"esp8266-0.02-1.bin\"},{\"version\":\"0.01-1\",\"file\":\"esp8266-0.01-1.bin\"}],\"ATmega\":[{\"version\":\"latest\",\"file\":\"atmega-0.01-1.hex\"},{\"version\":\"0.01-1\",\"file\":\"atmega-0.01-1.hex\"}]}";

JsonObject& root = jsonBuffer.parseObject(json);

for (JsonObject::iterator it=root.begin(); it!=root.end(); ++it)
{
    Serial.println("Iterate");
    if (!strcmp(it->key, "esp8266"))
    {
        const char* version = it->value["version"];
        Serial.println("version=");
        Serial.println(version);
        // ...
    }
}
```

bblanchon, 2015-11-24:
```c++
const char* version = it->value[0]["version"];
```

frippe75, 2015-11-24:
Worked perfectly! Thanks for the support

bblanchon, 2015-11-24:
You're welcome.
It was a tough one; people are often confused when the JSON contains mixed arrays and objects.

To get the number of version:

```c++
size_t count = it->value.size();
```

Thanks for using ArduinoJson.
Don't hesitate to edit the wiki.

frippe75, 2015-11-24:
Will try it!

Tried other options but failed.
Would have like the get that list as a node via:
```
JsonArray& esp8266 = root.getNodeAt("esp8266");
JsonObject& latest = esp8266.getNodeAt("latest");
```
But I guess it doesn't work that way. I would have like to traversed the nodes.

frippe75, 2015-11-24:
Thanks again! Problem solved!

Iterate
Found esp8266
ver=latest
filename=esp8266-0.02-1.bin
ver=0.02-1
filename=esp8266-0.02-1.bin
ver=0.01-1
filename=esp8266-0.01-1.bin
Iterate

bblanchon, 2015-11-24:
You can do that of course:

```c++
JsonArray& esp8266 = root["esp8266"];
```

But to get the latest you have to iterate or assume it's in the first slot.

closed 2015-12-01 by bblanchon