bblanchon/ArduinoJson #169: Question regarding iteration

susie-derkin, 2015-11-26:
Hi, 

I have a question how to iterate through a json string like this:
{
	"filestest":
	[
	  {"2index.html":1},
	  {"2style.css":2},
	  {"2favicon.ico":2}
	]

}

I want to iterate through that. When I use your example with an JsonArray I am not able to get the key or value member - the compiler says the JsonVariant it does not have a member key or value....

The example I used:
			for (JsonArray::iterator it=filestest.begin(); it!=filestest.end(); ++it)
			{
				const char* value = *it;   // not possible to get a member like key....
				Serial.println(value);
			}

Can anybody help to get the key and the value content in such an iteration?

Thanks in advance

bblanchon, 2015-11-26:
Hi,

`filetest` is an array of objects, so the iterator will point to an object.
Then if you want the name of the keys of theses nested object, you'll have to iterate again:

```c++
for (JsonArray::iterator i = filestest.begin(); i != filestest.end(); ++i) {
  JsonObject& nestedObject = *i;
  for (JsonObject::iterator j = nestedObject.begin(); j != nestedObject.end();
       ++j) {
    const char* key = it->key;
    int value = it->value;
  }
}
```

Regards,

susie-derkin, 2015-11-26:
MANY THANKS for your fast and absolutly helpful answer!
Now I get what I want!

Thank you Benoît!!!!!

bblanchon, 2015-11-27:
You're welcome.
This kind of question is very frequent.
People are often confused when there are mixed arrays and object.

I think there is either a lack of ergonomics or a lack of documentation.
What do you think?
Did you miss documentation material?
What would have help you find the answer by yourself?
Where did you look for before asking here?

susie-derkin, 2015-11-27:
Hi Benoît,

again thanks for your fast reply yesterday, for me you solved my question that I had over the last days!!

Regarding your questions:
I read the wiki articles in the wiki, -> Decoding and the Examples page and also the API reference and the FAQ, but those two were not so helpful. For me the examples on the first two pages were most useful, and I think some more examples with more complex JSON on the examples page would help me and other people very much!
I have to admit that I do not really have much experience with your very elegant programming style using templates and generic types <T> (it is not a good excuse but my C++ learning time was 20 years ago during my studies, in these times there was nothing like the STL....). So for me it is not clear from what type the iterator it (as yesterday) really is....
I also tried google, of course, but in that case it did not really help. And then I saw in the issues that you yourself will anser questions  -  and so I tried. Thanks again.

Anda nice evening for you!

closed 2015-12-22 by bblanchon