bblanchon/ArduinoJson #170: Unisgned long printed as long

pinps, 2015-12-01:
Hi,
I am trying to print an unsigned long value ID on Arduin Uno, but it seems to print it as a long:
Code:
void setup() {
  Serial.begin(9600);
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["id"] = 2345678901;
  root.printTo(Serial);
}
Output:
{"id":-1949288395}

Do I need to specify some how it is an unsigned or how to fix this behavior?

-Pinps

bblanchon, 2015-12-01:
Hi,

Internally, every value of an integral type is casted to `long`.
Currently, there is no plan to add support for `unsigned long`, because it'd increase the size of the code significantly.

Here are two workarounds:

```c++
root["id"] = double_with_n_digits(2345678901, 0);
root["id"] = ArduinoJson::Internals::Unparsed("2345678901");
```

Sorry about that.

bblanchon, 2015-12-01:
After thinking about it, I think I can change the internal storage to be `unsigned long` and add a variant type for negative value.
That way the impact on code size would be reasonable.

Anyway, I'm adding support for `long long` right now, I'll see how I can add this feature whe it's done.

pinps, 2015-12-03:
Thanks for workaround. The double_with_n_digits truncated last bits but unparsed worked great:
``` c
unsigned long id = 2345678901;
char buf[11];
sprintf(buf,"%lu",id);
root["id"] = ArduinoJson::Internals::Unparsed(buf);
```
Output now: {"id":2345678901}

bblanchon, 2015-12-04:
You may have to write:

```c++
root["id"] = ArduinoJson::Internals::Unparsed(jsonBuffer.strdup(buf));
```

if `buf` goes out of scope.

bblanchon, 2016-04-07:
I fixed the issue one month ago, but I was waiting for [this pull request](https://github.com/arduino/Arduino/pull/4667) to be merged.
Without this fix in the Arduino core, this feature will increase the size of the code.

closed 2016-04-28 by bblanchon