bblanchon/ArduinoJson #171: Support for long long

bblanchon, 2015-12-01:


bblanchon, 2016-02-14:
Fixed in 5.1.0

closed 2016-02-14 by bblanchon

cyberreefguru, 2016-02-16:
First - thanks so much for the amazing library. Totally rock solid! Kudos to you.

I just pulled the latest and tried to use use long long without any luck.  I get some very odd c++ error I've never seen before;

ArduinoJson/include/ArduinoJson/Internals/../JsonVariant.hpp:153:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableIf<false, ArduinoJson::JsonObject&>'
In file included from /ArduinoJson/include/ArduinoJson/JsonObject.hpp:153:0,
                 from /ArduinoJson/include/ArduinoJson/JsonArray.ipp:11,
                 from /ArduinoJson/include/ArduinoJson/JsonArray.hpp:181,
                 from /ArduinoJson/include/ArduinoJson.h:9,
                 from /ArduinoJson/ArduinoJson.h:14,
                 from ../Command.h:15,
                 from ../Command.cpp:8:
/ArduinoJson/include/ArduinoJson/JsonObject.ipp: In member function 'T ArduinoJson::JsonObject::get(ArduinoJson::JsonObjectKey) const [with T = long long int]':
/ArduinoJson/include/ArduinoJson/JsonObject.ipp:25:1: warning: control reaches end of non-void function [-Wreturn-type]

I looked in the source and could not find support for "long long" in the template of the "as" method.

Do I need to define "ARDUINOJSON_USE_LONG_LONG" or something like that?

Any clues what I'm missing?

-Tom

bblanchon, 2016-02-16:
Hi Tom,

You found a bug in the library or in your compiler (unlikely but possible).
Can you please open a new issue and specify the name, version and architecture of the compiler that you're using?

For your information, ArduinoJson is regularly tested on:

1. GCC 4.6.3
2. GCC 4.8.1
3. GCC 4.9.2
4. Clang 3.4.0
5. Clang 3.6.2
6. Apple Clang 7.0.2
7. Visual Studio 2010
8. Visual Studio 2012
9. Visual Studio 2013
10. Visual Studio 2010

Thanks for reporting this bug, this is of great help!

Regards.