bblanchon/ArduinoJson #172: How  to parse multi-level Json string

Pillar1989, 2015-12-02:
Hi 
     A very useful project. is there any docment about how to parse muti-level json ?
eg:
	  StaticJsonBuffer<200> jsonBuffer;
	  char json[] =
		  "{\"14\":{\"9\":\"1,#dfa,sdf#,1.89,*4\"}}";

	  JsonObject& root = jsonBuffer.parseObject(json);

	  if (!root.success()) {
		Serial.println("parseObject() failed");
		return;
	  }
	for (JsonObject::iterator it=root.begin(); it!=root.end(); ++it)
	{
	  Serial.println(it->key);
	  Serial.println(it->value.asString());
	}

I want to get all key and value, how to parse multi-level Json string ?

bblanchon, 2015-12-02:
Hi,

Here is a recursive function that will dump every key of every object in the tree:

```c++
void printObjectKeys(JsonObject& obj)
{
	for (JsonObject::iterator it=obj.begin(); it!=obj.end(); ++it)
	{
		Serial.println(it->key);
		
		if (obj->value.is<JsonObject&>())
		{
			printObjectKeys(obj->value);
		}		
	}
}
```

Of course, if the tree also contains arrays, you'll have to add another `if`.

Please check out the [examples in the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Examples).
Feel free to edit if you think you can improve it.

Pillar1989, 2015-12-02:
Thank you,
It should be like that:

void printObjectKeys(JsonObject& obj)
{
    for (JsonObject::iterator it=obj.begin(); it!=obj.end(); ++it)
    {
        Serial.println(it->key);

        if (it->value.is<JsonObject&>())
        {
            printObjectKeys(it->value);
        }
			else
			{
					Serial.println(it->value.asString());
			}
    }
}

closed 2015-12-02 by Pillar1989