bblanchon/ArduinoJson #175: possibility to know the type of JsonObject value

lapinroux, 2015-12-07:
Hello.
Is there a way to know easily if my JsonObject is a int or a string ? Because my object object["value"].as<long>(); returns '0' if there is a string (or anything different than a number) or if there is really 0 as int, then I don't know how to identify string or int.

The fast solution is to take everything as strnig and to use atoi or any number parsing tool, but I think that knowing this information directly when the value is read should be more beautiful. Something like object["value"].kindOfValue() == int

Very good library, go on like this.

bblanchon, 2015-12-07:
```c++
if (object["value"].is<int>()) {

}
```

lapinroux, 2015-12-08:
Really excuse me, I didn't find it on wiki. I close this issue.

closed 2015-12-08 by lapinroux

bblanchon, 2015-12-08:
You're welcome.

The documentation is incomplete and there are definitely some discoverability issues.
I'm working on that.

Thanks for using ArduinoJson.