bblanchon/ArduinoJson #176: memory management ....

SiFly, 2015-12-08:
From what I can tell all usages of the StaticJsonBuffer<X> must be on the stack

If you try and have a static StaticJsonBuffer<> and re use it many times then you are in fact in trouble as it runs out of space and your json stops working. 

Can you please add a reset method to the StaticJsonBuffer template such that once you have finished with your json you can use it again

bblanchon, 2015-12-08:
Please read [FAQ: How to reuse a JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-reuse-a-jsonbuffer).

closed 2015-12-08 by bblanchon

SiFly, 2015-12-08:
Hello !

Thanks for your prompt reply

I was just trying to keep to an absolute  limit the stack usage on the
arduino ... and thought having a static one of these would be nice to have
since the arduino compiler does not track stack space used by your program
but it does keep track of static variables which is useful

Of course what you could do us track the objects allocated and if all are
out of scope you reset the primary allocator back ... but you probably do
not want to do that !

I've read your comments and understand your concerns about use of variables
which go out of scope ...  but this is a common pitfall in software ... and
maybe you could out a warning on a putative reset/clear method if you
choose to implement ... just to say here be dragons ....

Simon