bblanchon/ArduinoJson #177: jsonBuffer size and parseObject() failed

roccomuso, 2015-12-11:
Hi,

Why this code gives me a parseObject() failed error on Serial Monitor, what am i doing wrong?

    String mex = "{\"cmd\": \"self_advertise\"}";

    const int BUFFER_SIZE = JSON_OBJECT_SIZE(1);
    StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
    
    JsonObject& root = jsonBuffer.parseObject(mex);

     if (!root.success()){
        Serial.println("ArduinoJson parseObject() failed");
     }

bblanchon, 2015-12-11:
Hi,

This is because `parseObject()` needs to modify the string.
Since you provide a `String`, which cannot be modified, it has to make a copy.

To fix this, either:

1. Increase the size of the `StaticJsonBuffer` by the number of characters in the string, or
2. replace the `String` by a `char[]`

Please read [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

roccomuso, 2015-12-12:
Thank you. Fixed using char[].

bblanchon, 2015-12-12:
Yes, that's the most efficient choice.
Programmers coming from C# or Java usually want to use the `String` class, but most of the time it's a mistake.

closed 2015-12-12 by bblanchon