bblanchon/ArduinoJson #178: Size of generated json

raimohanska, 2015-12-12:
Hi!

I'm planning to use ArduinoJson for POSTing some JSON data on a server. I'm sure I can print the json content to my `client` like this:

```c
JsonObject& root = jsonBuffer.createObject();
// generate the content here
root.printTo(client);
```

However, oo be able to do this correctly, I need to know the size of the generated JSON so that I can populate the `Content-Length` header correctly. Is there an easy way to get the size of the JSON data that is generated using the `printTo` method?

raimohanska, 2015-12-12:
Oh, I guess this is the `measureLength` method. I'll try that!

We might wanna add that method to the documentation?

raimohanska, 2015-12-12:
I added my 5 cents to the Wiki here: https://github.com/bblanchon/ArduinoJson/wiki/Encoding-JSON

raimohanska, 2015-12-12:
Thanks for the great library! I'm using it in my IoT hobby project here: https://github.com/raimohanska/esp-dht-transmitter

closed 2015-12-12 by raimohanska

bblanchon, 2015-12-13:
Hi Juha,

I'm glad you found the answer!
Thank you very much for editing the wiki :wink:
Please add you project to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).