bblanchon/ArduinoJson #179: Parsing failure

cawena, 2015-12-17:
the following json will not parse:
{"host":"http://transport.opendata.ch","apiVersion":"/v1","resources":"/connections","from":"Kräyigen","to":"bern","limit":"2","fields":"connections/from/departure"}

I located the problem on value "connections/from/departure". When I remove this key and value, the parsing is successful. I also tried "connections\/from\/departure", but thats also not working.

domonetic, 2015-12-18:
Check size of jsonbuffer

cawena, 2015-12-18:
to small? I have just five objects;
{"host":"http://transport.opendata.ch","apiVersion":"/v1","resources":"/connections","from":"Kräyigen","to":"bern","limit":"2","fields":"connections/from/departure"}

cawena, 2015-12-18:
I also tried to escaping the slashes

char json[] = "{\"host\":\"http:transport.opendata.ch\",\"apiVersion\":\"v1\"}";
return
http:transport.opendata.ch
v1

char json[] = "{\"host\":\"http:\/\/transport.opendata.ch\",\"apiVersion\":\"\/v1\"}";
Retrun

/v1
-> host is missing

bblanchon, 2015-12-18:
@cawena Please copy/paste your program here.

cawena, 2015-12-18:
I use Arduino Yun:
content of config.json
{"url":{"host":"http://transport.opendata.ch","apiVersion":"/v1","resources":"/connections","from":"Kräyigen","to":"bern","limit":"2","fields":"connections/from/departure"}}

it was nicer to have the json in a pretty format.

like this: :+1: 
{
	"url": {
		"host": "http://transport.opendata.ch",
		"apiVersion": "/v1",
		"resources": "/connections",
		"from": "Kräyigen",
		"to": "bern",
		"limit": "2",
		"fields": "connections/from/departure"
	}
}

anyway, here the code. just the void.

void getConfig() {
  String jsonStr;
  File myFile = FileSystem.open("/mnt/sda1/config.json", FILE_READ);

  if (myFile) {
    while (myFile.available()) {
      char c = myFile.read();
      jsonStr += c;
    }
  } else {
    // showErrorOnMatrix();
  }

  StaticJsonBuffer<200> jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(json);
  if (!root.success()) {
    Serial.println("Parsing fail");
    return;
  }

  JsonObject& url = root["url"];
  String host = url["host"];
  String api = url["apiVersion"];
  String resources = url["resources"];
  String stationFrom = url["from"];
  stationFrom = urlDecode("from=" + stationFrom);
  String stationTo   = url["to"];
  stationTo = urlDecode("to=" + stationTo);
  String limit = url["limit"];
  limit = urlDecode("limit=" + limit);
  String fields = root["fields"];
  fields = urlDecode("fields[]=" + fields);

 p.begin("curl");
  p.addParameter("-H \"Cache-Control: no-cache\"");
  p.addParameter(host + apiVersion + resources + "?" + stationFrom + "&" + stationTo + "&" + limit + "&" + fields);
  p.run();

String jsonData = "";
  while (p.available() > 0) {
    char c = p.read();
    jsonData += c;
  }

THE REST IS NOT DONE YET.

}

bblanchon, 2015-12-18:
The problem is that you use a `String` as the input of the parser.
In this situation, the `JsonBuffer` makes a copy of the string so it can modify it.
See [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails).

There are 4 workarounds, from best to worst:

1. Stop using a `String`
2. Allow the `String` to be modified by the parser, by casting it to `char*`
3. Increase the size of the `StaticJsonBuffer` so that the JSON input fits into it
4. Switch to a `DynamicJsonBuffer` that will grow as needed

Here is how to implement solution 2:

```c++
JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(json.c_str()));
```

Note that the content of the `json` variable will be modified.

cawena, 2015-12-18:
OK, then maybe it's better to read the file content in a char array to process it better.
OK thx for the hint.

closed 2015-12-29 by bblanchon

pkourany, 2016-02-23:
@bblanchon, I'm working on the Particle Photon with what I believe is version 4 (or so) of a[ ported ArduinoJson library](https://github.com/menan/SparkJson).  I am receiving an http response into a string (response.body in a global structure), as the OP did, and would like to parse it.  The json string is:

{"UID":"420023001747343337363432","AID":0,"VER":"1.0","CFGRX":0,"ECHO":1,"KEYS":"Date","A0","A1"],"VALUES":[[1456199015,2,4],[1456199015,1,5]]}

I tried copying the String to a char array and parsing but that failed:
```
StaticJsonBuffer<500> jsonReply;
strncpy(jresp, response.body.c_str(), sizeof(jresp));  //char jresp[300];
JsonObject& root = jsonReply.parseObject(jresp);  // 
```

I also tried casting as you described and that failed:
```
JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(response.body.c_str()));
```
The only thing I did not try is using a DynamicJsonBuffer.  Any thoughts?

bblanchon, 2016-02-23:
@pkourany The JSON input is invalid.

This is pretty obvious with syntax highlighting: 

```json
{"UID":"420023001747343337363432","AID":0,"VER":"1.0","CFGRX":0,"ECHO":1,"KEYS":"Date","A0","A1"],"VALUES":[[1456199015,2,4],[1456199015,1,5]]}
```

pkourany, 2016-02-23:
@bblanchon, that's my fault - it's a cut and paste error.  The correct (and valid) json string is:
```
{"UID":"420023001747343337363432","AID":0,"VER":"1.0","CFGRX":0,"ECHO":1,"KEYS":["Date","A0","A1"],"VALUES":[[1456199015,2,4],[1456199015,1,5]]}
```

bblanchon, 2016-02-23:
Just to be sure, can you run this:

```c++
StaticJsonBuffer<500> jsonBuffer;
Serial.println(response.body.c_str());
JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(response.body.c_str()));
Serial.println(root.success());
```

I really what to see the actual input, because most of the time it's the source of the trouble.

pkourany, 2016-02-24:
@bblanchon, here is the output of the code:
```
{"UID":"420023001747343337363432","AID":0,"VER":"1.0","CFGRX":0,"ECHO":1,"KEYS":["Date","A0","A1"],"VALUES":[[1456272595,2,3],[1456272595,2,6]]}
0
```
Same as posted previously.

pkourany, 2016-02-24:
@bblanchon, I just substituted the StaticJasonBuffer<500> with DynamicJasonBuffer and the parse worked.  I guess 500 is not large enough!  Thanks!

bblanchon, 2016-02-25:
@pkourany I don't think so. Most likely the stack is full, so using the heap solved your issue.

pkourany, 2016-02-25:
@bblanchon, ah yes, the stack.  I should know better!

uzbhutta, 2016-05-25:
Not sure if it's appropriate to address this in such an old thread, but like pkourany, I am using the Particle Photon. I am too receiving an http response into a string (which cannot be avoided given the nature of Particle's workflow). More specifically, through the Particle workflow, I pass in a string of JSON into the outletToggle function below. However, I always receive an error while parsing. 

```
int outletToggle(String input) {
  StaticJsonBuffer<64> jsonBuffer;

  //char json[] = "{\"sensor\":\"mic\"}";
  //JsonObject& root = jsonBuffer.parseObject(json);

  JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(input.c_str()));

  if (!root.success()) {
    Serial.println("Error parsing input.");
  }

  Serial.println(root.success());
  const char* jsonParsed = root["sensor"].asString();
  Serial.println(jsonParsed);
}
```

If I comment out the line `  JsonObject& root = jsonBuffer.parseObject(const_cast<char*>(input.c_str())); ` , and uncomment the two lines about it, the parsing occurs correctly and jsonParsed outputs "mic" as expected. However, passing something into the function presents problems. 

In fact, even if I use the following, parsing works fine. I believe the problem has to do with the passing into the function of the string. 

```
  String json1 = "{\"sensor\":\"gps\"}";
  char* json = const_cast<char*>(json1.c_str());
  JsonObject& root = jsonBuffer.parseObject(json);
```

Any help would be appreciated. Thanks!

bblanchon, 2016-05-25:
@uzbhutta This function looks OK to me. Is there any chance that the input string contains invalid JSON, or that the `StaticJsonBuffer` is too small?

BTW, since you're using a `String` as a parameter, and not a `String&`, an implicit copy is made.
If that's really what you want, you should rewrite the function this way, as it would avoid the `const_cast`:

```c++
void outletToggle(const String& input) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(input); // <- a copy of the String is made

  if (!root.success()) {
    Serial.println("Error parsing input.");
    return;
  }

  const char* jsonParsed = root["sensor"];
  Serial.println(jsonParsed);
}
```

Have a look at:

* [Avoiding Pitfalls: Make StaticJsonBuffer big enough](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#1-make-staticjsonbuffer-big-enough)
* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

uzbhutta, 2016-05-26:
@bblanchon I've triple checked and the JSON string is valid and the StaticJsonBuffer is adequate. I'm not quite sure why the function isn't working when 'input' is passed in, even though it works if it's declared inside the function. 

As for your suggestion, I'm using Particle's official IDE built on Atom, called Particle Dev. Unfortunately, whenever I try and avoid the const_cast and just pass in a String as a parameter in parseObject(), I get the error "invalid user-defined conversion from 'const String' to 'char*' [-fpermissive].  

I appreciate your help in advance.

bblanchon, 2016-05-26:
> invalid user-defined conversion from 'const String' to 'char*'

This is very strange because there is and overload taking a `const String&`.
I see only two possible reasons for this error:

1. You're using ArduinoJson 4.x, which had no support for `String`
2. You're using a different class named `String`. In that case try `std::string` or `ArduinoJson::String`.

uzbhutta, 2016-05-26:
@bblanchon Seems to me as if I was using an older version. All the cpp files in my version only have 2014 as their top comment. I'll update the library and try again.

If you don't mind me asking, what would be the downsides of having to do the const_cast? Is there any reason you suggest avoiding it?

bblanchon, 2016-05-26:
`const_cast` is a keyword that yells "I'm doing something dangerous here!".
It's the kind of things you try to avoid because it disables the safeties.

That being said, and since you state that you have to use a `String`, the `const_cast` is the only option to avoid a duplication of the string.

However, as I pointed out in the first message, you're doing a duplication anyway.
So, we can avoid the always-disturbing `const_cast` keyword.