bblanchon/ArduinoJson #180: asString problem for a valid json

bmafy, 2015-12-22:
Hi, I find your library very useful and quite easy to use, but now I'm facing an annoying issue converting my json (generated using your method) in a string and nothing happens. The converted string (json-->string) is void. Are there any "good methods" for using asString? I didn't find anything excpeting the doc in the wiki page. Thank you very much for your help.
Matteo

bblanchon, 2015-12-22:
Hi Matteo,

Can you please copy/paste you program here?

Benoit

closed 2015-12-29 by bblanchon

bmafy, 2016-01-29:
Hi! Sorry for delayed answer but it turns out that I was wrong doing in that way... A simple:
`JsonObject.printTo(buffer, sizeof(buffer));`
figured it :+1: 
Thanks for your help!!

bblanchon, 2016-01-29:
Hi Matteo,
I'm glad you found the answer.
Please apologize for closing the issue without confirmation but, since you gave no response in 7 days, I assumed you moved on.
Thanks for using ArduinoJson.

bmafy, 2016-01-30:
No problem at all, I just wanted to let other people know what my solution has been.