bblanchon/ArduinoJson #181: Platformio compatibility?

sticilface, 2015-12-22:
Do you have any experience of getting this to work with platformio?

I followed these instructions
http://www.penninkhof.com/2015/12/1610-over-the-air-esp8266-programming-using-platformio/

and it works flawlessly.. I copied all my libs to the /lib folder but when i try and compile them it is the arduinojson that stops it in its tracks.  It is the `#include "../include/ArduinoJson.h"` that causes the problems.  The one included for ardunio compatibility... commenting it out, stops it from working..  bit of a stab in the dark asking you, but my knowledge of compiling structures / directories, outside the arduino IDE is fairly minimal. any help appreciated

bblanchon, 2015-12-22:
Can you please copy/paste the compilation error?

sticilface, 2015-12-22:
```
======================================================================== [ ERROR ] Took 5.70 seconds ========================================================================
Andrew-MBA:melvana amelvin$ platformio run --target upload
[Tue Dec 22 15:47:42 2015] Processing nodemcu (platform: espressif, upload_port: 192.168.5.55, board: nodemcu, framework: arduino, board_f_cpu: 160000000L)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/src/Melvana.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=160000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -DPLATFORMIO=020603 -I/Users/amelvin/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant -I.pioenvs/nodemcu/ESP8266WiFi -I.pioenvs/nodemcu/ESP8266WebServer -I.pioenvs/nodemcu/ArduinoOTA -I.pioenvs/nodemcu/ArduinoJson -I.pioenvs/nodemcu/ESP8266mDNS -I.pioenvs/nodemcu/ESP8266HTTPUpdateServer -I.pioenvs/nodemcu/ESP8266HTTPClient -I.pioenvs/nodemcu/ESPmanager -I.pioenvs/nodemcu/NeoPixelBus -I.pioenvs/nodemcu/Adafruit-GFX-Library-master -I.pioenvs/nodemcu/GDBStub -I.pioenvs/nodemcu/SimpleTimer-master -I.pioenvs/nodemcu/FSBrowser src/Melvana.cpp
In file included from .pioenvs/nodemcu/ESPmanager/ESPmanager.h:38:0,
from src/Melvana.h:5,
from src/Melvana.cpp:1:
.pioenvs/nodemcu/ArduinoJson/ArduinoJson.h:13:36: fatal error: ../include/ArduinoJson.h: No such file or directory
#include "../include/ArduinoJson.h"
^
compilation terminated.
```

so i actually got somewhere, if i use 
```
#include <include/ArduinoJson.h> 
```

I don't get the error, but it fails on one of my functions later.. I guess though it is not 'drop in' compatible with platformio using arduino

sticilface, 2015-12-22:
ah... just found an issue with platformio.... https://github.com/platformio/platformio/issues/357

sorry. thought i searched there!

sticilface, 2015-12-22:
Thought i made loads of progress... got past that.. and another error and thought i was on my way... then...

```
nvs/nodemcu/GDBStub -I.pioenvs/nodemcu/SimpleTimer-master -I.pioenvs/nodemcu/FSBrowser .pioenvs/nodemcu/ArduinoJson/test/ArduinoString_Tests.cpp
.pioenvs/nodemcu/ArduinoJson/test/ArduinoString_Tests.cpp:7:25: fatal error: gtest/gtest.h: No such file or directory
#include <gtest/gtest.h>
^
compilation terminated.
scons: *** [.pioenvs/nodemcu/ArduinoJson/test/ArduinoString_Tests.o] Error 1
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/ArduinoJson/test/DynamicJsonBuffer_Array_Tests.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=160000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=200 -I/Users/amelvin/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant -I.pioenvs/nodemcu/ESP8266WiFi -I.pioenvs/nodemcu/ESP8266WebServer -I.pioenvs/nodemcu/ArduinoOTA -I.pioenvs/nodemcu/ArduinoJson -I.pioenvs/nodemcu/ESP8266mDNS -I.pioenvs/nodemcu/ESP8266HTTPUpdateServer -I.pioenvs/nodemcu/ESP8266HTTPClient -I.pioenvs/nodemcu/ESPmanager -I.pioenvs/nodemcu/NeoPixelBus -I.pioenvs/nodemcu/Adafruit-GFX-Library-master -I.pioenvs/nodemcu/GDBStub -I.pioenvs/nodemcu/SimpleTimer-master -I.pioenvs/nodemcu/FSBrowser .pioenvs/nodemcu/ArduinoJson/test/DynamicJsonBuffer_Array_Tests.cpp
.pioenvs/nodemcu/ArduinoJson/test/DynamicJsonBuffer_Array_Tests.cpp:7:25: fatal error: gtest/gtest.h: No such file or directory
#include <gtest/gtest.h>
^
compilation terminated.
scons: *** [.pioenvs/nodemcu/ArduinoJson/test/DynamicJsonBuffer_Array_Tests.o] Error 1
======================================================================== [ ERROR ] Took 17.89 seconds ========================================================================
```

sticilface, 2015-12-22:
ok, i think my approach of putting the original github src folders into lib is the mistake... its compiling all the examples!  need to read more

bblanchon, 2015-12-22:
You should use the zip packages instead of making a git clone, because the packages don't include the tests.
Then, please don't `#include <include/ArduinoJson.h>`, you should really write `#include <ArduinoJson.h>`.
Why not moving `lib/ArduinoJson/src/Arduino.h` to `lib/ArduinoJson/Arduino.h` and modify the file accordingly?

ivankravets, 2015-12-22:
@bblanchon sorry for this issue. Please merge my PR https://github.com/bblanchon/ArduinoJson/pull/182

P.S: This is temporary solution until https://github.com/platformio/platformio/issues/289 has not been fixed yet.

ivankravets, 2015-12-22:
Temporary solution is to use `#include <include/ArduinoJson.h>`

P.S: @sticilface don't forget to `platformio lib update`

sticilface, 2015-12-22:
Hi,  thanks for the speedy work. 
I've done both these things, and I'm still getting the multiple builds msg...
```
=========================================================================== [ ERROR ] Took 1.49 seconds ===========================================================================
Andrew-MBA:melvana amelvin$ platformio run
[Tue Dec 22 18:53:27 2015] Processing nodemcu (platform: espressif, upload_port: 192.168.5.55, board: nodemcu, framework: arduino, board_f_cpu: 160000000L)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

scons: warning: Two different environments were specified for target /Users/amelvin/platformio/melvana/.pioenvs/nodemcu/libJson_ID64.a,
but they appear to have the same action: $AR $ARFLAGS $TARGET ${_huge_sources_hook(SOURCES)}
$RANLIB $RANLIBFLAGS $TARGET
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 203, in BuildLibrary

scons: *** Multiple ways to build the same target were specified for: /Users/amelvin/platformio/melvana/.pioenvs/nodemcu/libJson_ID64.a  (from ['/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Arduino/Print.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/Comments.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/Encoding.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/IndentedPrint.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/JsonParser.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/List.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/Prettyfier.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/Internals/StaticStringBuilder.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/JsonArray.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/JsonBuffer.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/JsonObject.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/JsonVariant.o'] and from ['/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Arduino/Print.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/Comments.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/Encoding.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/IndentedPrint.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/JsonParser.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/List.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/Prettyfier.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/Internals/StaticStringBuilder.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/JsonArray.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/JsonBuffer.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/JsonObject.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/src/JsonVariant.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/ArduinoString_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/DynamicJsonBuffer_Array_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/DynamicJsonBuffer_Basic_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/DynamicJsonBuffer_NoMemory_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/DynamicJsonBuffer_Object_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/GbathreeBug.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/IntegrationTests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/Issue10.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/Issue34.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/Issue67.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/Issue90.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Add_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Basic_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Invalid_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Iterator_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_PrettyPrintTo_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_PrintTo_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Remove_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Set_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonArray_Subscript_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonObject_Basic_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonObject_ContainsKey_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonObject_Invalid_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/JsonObject_Iterator_Tests.o', '/Users/amelvin/platformio/melva [... cut ...] ts.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/StaticJsonBuffer_ParseArray_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/StaticJsonBuffer_ParseObject_Tests.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/StdStream.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/Json_ID64/test/StringBuilderTests.o'])
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 203, in BuildLibrary
```

bblanchon, 2015-12-22:
@ivankravets :

> Temporary solution is to use `#include <include/ArduinoJson.h>`

This is driving me mad :frowning:
Why don't you add the `include/` folder to the search list?

ivankravets, 2015-12-22:
@sticilface you use obsolete version of library. I've asked above to run this command `platformio lib update`

sticilface, 2015-12-22:
i have
[image]

but it still give the same error!

ivankravets, 2015-12-22:
@bblanchon the problem is linked with philosophy of PlatformIO Code Builder and Arduino IDE behaviour. PlatformIO doesn't use original location of the lib files to build source code. It symlink/clone to `.pioenvs` temporary folder and build them. It tries to find `include` in the root of the library. Then if sees `src` directory tries to find here. If it found, then `src` directory is nominated as `root` directory of the library.  In your case, you try to reach "includes" which are located behind `src` directory. This is directory isn't symlinked/clonned to `.pioenvs`.

Ok, I'll try to improve compatibility with Arduino IDE behaviour.

ivankravets, 2015-12-22:
@sticilface do you have this library in other lib locations? Like in project's `lib` dir?

sticilface, 2015-12-22:
no.  I did but I've deleted it.  
```
platformio run --target clean
```
gives error as well

ivankravets, 2015-12-22:
@sticilface paste here your sketch. Thanks.

sticilface, 2015-12-22:
right ok... my bad... this is a huge sketch with all sorts of complexities....  so... i made a simply sketch and it does not produce this error...  it does however produce the gtest error...  i went back... and found that i've included arduinojson in another location and had not put <include/arduinojson.h>...

where are the libs downloaded to on mac.. so i can delete the gtest?

bblanchon, 2015-12-22:
@ivankravets 

> In your case, you try to reach "includes" which are located behind src directory

That's why I told you to exclude `src/ArduinoJson.h`, because this file makes no sense in this environment.

sticilface, 2015-12-22:
found the lib64 install location.  deleted test.. and I'm in business... still would like to fix it so the ino files are drop in compatible with arduino..

sticilface, 2015-12-22:
GDBStub does not work though

ivankravets, 2015-12-22:
> That's why I told you to exclude src/ArduinoJson.h, because this file makes no sense in this environment.

As I've said, PlatformIO analyses all includes. The one solution in this structure is to create the same "hook"(as for Arduino IDE) for PlatformIo and place it to the root of project. Do you have any objections?

``/ArduinoJson.h``
```c
#include "include/ArduinoJson.h"
```

Thanks!

ivankravets, 2015-12-22:
@sticilface I don't understand what are you doing :disappointed: Where did you find `test` and etc dirs? See the latest version of the lib http://platformio.org/#!/lib/show/64/Json. Try to download it manually in this page. Where is `test` folder?

bblanchon, 2015-12-22:
@ivankravets 

> The one solution in this structure is to create the same "hook"(as for Arduino IDE) for PlatformIO and place it to the root of project

You realize that I cannot do that for every platform?
I have no objection to have a dedicated package, but the problem with platform.io is that you get the files from the github repo, so I don't have the opportunity to remodel the source tree.

Also, I bet that if I add this file in the source, it will break some builds on Arduino.

sticilface, 2015-12-22:
> @sticilface I don't understand what are you doing  Where did you find test and etc dirs? See the latest version of the lib http://platformio.org/#!/lib/show/64/Json. Try to download it manually in this page. Where is test folder?

Haha. Sometimes I don't understand either. However on this occasion I do!!  

The tests folder is included when you simply use 

Platformio lib install 64 

That is it. I found the tests folder in the libs64 folder. So it is downloaded along with everything else when you use platformio to install the lib! 

I can try deleting it and redownload! Not at the computer at the moment. 

Any pointers towards getting gdbstub working?

ivankravets, 2015-12-22:
The problem here is Arduino IDE. They invented a few "bicycles" like 
* "*.ino/*.pde"
* if no `src` directory in the root, DON'T build nested source files. Only the files from root dir should be built
* if `src` exists, build ONLY from src.
* if is dir `/src/utility` or `utility` it should be added to `CPPPATH` automatically.
* etc.etc.etc.

I don't know why we should count that this is RIGHT project structure. PlatformIO has support for approximately 200 embedded boards from more than 15 different development platforms. We can say "ARM MBED" community that we should keep project/library structure as Arduino likes it.

I don't have any objections for Arduino IDE. I think about users or even beginners. Arduino IDE with these "bicycles" breaks common approach for C++ development (someone convert INO to CPP and generates prototypes "on-the-fly",....).

Let's resolve this issue from users point. You made great library, people want to use it. From my point of view, if you place `ArduinoJson.h` - this is RIGHT solution. I mean, this isn't a hook. Hook is `src/ArduinoJson.h`. However, you say that it will break compatibility with Arduino.

What I can propose:

1. [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) allows to specify `url` and `version` fields for the lib.
2. I can clone you repo and place this `ArduinoJson.h` to the root of project
3. You can create separate branch in your project and make step 2.

In both cases we have option:
1. To use the latest version "as the latest commit" (default behaviour)
2. To fix library to the specific source code and library version.

WDYT?

ivankravets, 2015-12-22:
@sticilface please remove `Json_ID64` from `~/.platformio/lib`, project's `lib` folders and install it via `platformio lib install 64`.

bblanchon, 2015-12-22:
@ivankravets I agree with you that the library layout of Arduino is really bad, and the "new" layout isn't reallly better.
However this library is called ArduinoJson, the least I can do is make sure it works on Arduino!
Give me a sec, I'll to try to move the hook to the root and see what happens with Arduino and Stino.

ivankravets, 2015-12-22:
@bblanchon thanks! I've just cloned your repo to https://github.com/ivankravets/ArduinoJson. But I can remove it if we resolve this issue without forks.

bblanchon, 2015-12-22:
Arduino is pretty clear:

    fatal error: ../include/ArduinoJson.h: No such file or directory

But isn't there an `include` option in `library.json` that we could use for that ?

ivankravets, 2015-12-22:
Currently [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) specifies how library manager should extract source code from repository/CVS, URL, etc. If I got you, you mean "options for builder". See https://github.com/platformio/platformio/issues/289 where we are going to allow developers to control our builder. It will be huge update. In this case, developer will have ability to customize own library beginning from extracting process and finishing with building.

> However this library is called ArduinoJson, the least I can do is make sure it works on Arduino!

I agree with you. You need to keep compatibility with Arduino IDE. PlatformIO this is my problem. Do you have any objections if I'll resolve this issue via https://github.com/ivankravets/ArduinoJson ?

ivankravets, 2015-12-22:
I've just wanted to test header file in root dir but can't build this lib with the latest Arduino IDE
```
"/Volumes/SOFT/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Volumes/SOFT/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Volumes/SOFT/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "-I/Users/ikravets/Documents/Arduino/libraries/ArduinoJson/src" "/var/folders/nj/nt38kxwd2xxfl5l42np_w7qw0000gn/T/build35bacdd26bf81b88c61d5742e8a33a78.tmp/sketch/JsonGeneratorExample.ino.cpp" -o "/var/folders/nj/nt38kxwd2xxfl5l42np_w7qw0000gn/T/build35bacdd26bf81b88c61d5742e8a33a78.tmp/sketch/JsonGeneratorExample.ino.cpp.o"
JsonGeneratorExample:9: error: 'ArduinoJson' does not name a type
 void setup() {

 ^
Using library ArduinoJson at version 5.0.7 in folder: /Users/ikravets/Documents/Arduino/libraries/ArduinoJson 
exit status 1
'ArduinoJson' does not name a type
```

I didn't edit example or any files.

bblanchon, 2015-12-22:
Are you using Arduino 1.6.7 ?

Don't worry, I made up my mind, I'll include yet another `ArduinoJson.h` in the source tree.
Platform.io looks really great, and deserve a little sacrifice :wink:

[I'm adding it to the travis build, I'm almost there](https://travis-ci.org/bblanchon/ArduinoJson/builds/98405327):

```
src/JsonParserExample.cpp: In function 'void setup()':
src/JsonParserExample.cpp:10:3: error: 'Serial' was not declared in this scope
Serial.begin(9600);
```

What file shall I include?

ivankravets, 2015-12-22:
> Are you using Arduino 1.6.7 ?

1.6.6 :-1: Sorry, let me update it. Is this problem with 1.6.6?

ivankravets, 2015-12-22:
> What file shall I include?

I'm working on your log...

bblanchon, 2015-12-22:
Yes, version 1.6.6 didn't allow a library to be in a namespace.

Can you have a look at my [travis script](https://github.com/bblanchon/ArduinoJson/blob/platformio/scripts/travis/platformio.sh) and tell me if I'm doing it right?
I saw [on the documentation](http://docs.platformio.org/en/latest/ci/travis.html) that there is a `platformio ci`. Should I use that instead ?

ivankravets, 2015-12-22:
> I saw on the documentation that there is a platformio ci. Should I use that instead ?

Sure, it's shortcut for all your commands :smile: . Try this

```sh
#!/bin/sh -eux

ARDUINO_JSON=$PWD

pip install --user --upgrade pip setuptools
pip install --user platformio

rm -r $ARDUINO_JSON/test

platformio ci $ARDUINO_JSON/examples/JsonParserExample/JsonParserExample.ino -l $ARDUINO_JSON -b uno
```

bblanchon, 2015-12-22:
It works now.
I added a loop to test several examples:

```
for EXAMPLE in JsonParserExample JsonGeneratorExample
do
	platformio ci examples/$EXAMPLE/$EXAMPLE.ino -l '.' -b $BOARD
done
```

Is it possible to download toolchain and framework only once ?

ivankravets, 2015-12-22:
> Is it possible to download toolchain and framework only once ?

What do you mean? Do you want to cache PlatformIO packages between builds? Please take a look on 
https://docs.travis-ci.com/user/integration/platformio/

```
# Cache PlatformIO packages using Travis CI container-based infrastructure
sudo: false
cache:
    directories:
        - "~/.platformio"
```

ivankravets, 2015-12-22:
How about to test the rest of boards http://platformio.org/#!/boards? Maybe, `esp01`, `due`, `teensy31`? They use different toolchains and frameworks.

bblanchon, 2015-12-22:
I'm sorry, since the build was quite long I assumed that the toolchain was downloaded twice.

ivankravets, 2015-12-22:
I recommend to add this to `.travis.yml`
```yml
# Cache PlatformIO packages using Travis CI container-based infrastructure
sudo: false
cache:
    directories:
        - "~/.platformio"
```

Travis will cache packages between builds.

ivankravets, 2015-12-22:
Thanks! Looks like it works good? :blush: This should be great present for our users :gift:

bblanchon, 2015-12-22:
You're welcome.
Congratulation for platform.io, i'm amazing how easy the integration has been.

closed 2015-12-22 by bblanchon

ivankravets, 2015-12-22:
Thanks a lot for this time!

P.S: Could you replace here https://github.com/bblanchon/ArduinoJson/blob/master/ArduinoJson.h#L9 to `PlatformIO`? `platform.io` this is Advertising Platform :smile: See http://platform.io

We are http://platformio.org

sticilface, 2015-12-22:
Great work guys, thanks for the speed on solving this.  44 comments and an issue closed in 7hours, fantastic!!!!!  Thank you

ivankravets, 2015-12-22:
@sticilface thanks that pinned us :blush: Could you re-test it? The library has been updated and work excellent! http://platformio.org/#!/lib/show/64/Json

I've retested it.

sticilface, 2015-12-22:
retested and working :)