bblanchon/ArduinoJson #184: Add object as String with 'firstString + secondString' fails

beegee-tokyo, 2015-12-23:
IDE: Arduino IDE
IDE version 1.6.6
Target: ESP8266
ArduinoJson version 5.0.7

First: Thanks a lot for this awesome library!!!! It is a great help to decode and encode JSON objects and arrays!

Second: Problem
I am trying to generate a json object with an entry called "result". The content of "result" would be a keyword (firstString) and a description (secondString).

When I try:
```javascript
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
...
String req = httpClient.readStringUntil('\r');
...
root["result"] = "fail " + req;
```
I get a long error list in Arduino IDE
```javascript
In file included from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.ipp:10:0,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.hpp:151,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson.h:9,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from D:\ESP8266\Projects\IR-FujiDenzo\IR-FujiDenzo.ino:23:

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp: In instantiation of 'ArduinoJson::JsonObjectSubscript<TKey>& ArduinoJson::JsonObjectSubscript<TKey>::operator=(const T&) [with T = StringSumHelper; TKey = const char*]':

D:\ESP8266\Projects\IR-FujiDenzo\IR-FujiDenzo_1_Subs.ino:253:18:   required from here

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:35:25: error: no matching function for call to 'ArduinoJson::JsonObjectSubscript<const char*>::assign(const StringSumHelper&)'

         JsonVariant>(src);

                         ^

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:35:25: note: candidate is:

In file included from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArraySubscript.hpp:9:0,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArray.ipp:10,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonArray.hpp:135,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson.h:8,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from D:\ESP8266\Projects\IR-FujiDenzo\IR-FujiDenzo.ino:23:

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonSubscriptBase.hpp:76:23: note: template<class TValue> TImpl& ArduinoJson::JsonSubscriptBase<TImpl>::assign(TValue) [with TValue = TValue; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]

   FORCE_INLINE TImpl& assign(TValue value) {

                       ^

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonSubscriptBase.hpp:76:23: note:   template argument deduction/substitution failed:

In file included from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.ipp:10:0,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.hpp:151,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson.h:9,

                 from D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from D:\ESP8266\Projects\IR-FujiDenzo\IR-FujiDenzo.ino:23:

D:\Arduino-Dev\Projects\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp:35:25: note:   cannot convert 'src' (type 'const StringSumHelper') to type 'ArduinoJson::JsonVariant'

         JsonVariant>(src);

                         ^

exit status 1
Error compiling.
```

If I use
```javascript
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
...
String req = httpClient.readStringUntil('\r');
...
String statResponse = "fail " + req;
root["result"] = statResponse;
```
it works fine.

Any possibility to implement 'String Addition Operator' functionality? I am running out of memory in my project and would not like to add another String variable.
If it is puts too much overhead on the library, I understand and will stick with a String variable to create the JSON entry.

bblanchon, 2015-12-23:
First, how did you manage to use ArduinoJson with Arduino 1.6.6? I thought it was impossible.

Second, the best way to waste memory is to use the `String` class, but I digress...

This bug will be quite difficult to solve because this `StringSumHelper` is yet another non-standard thing of Arduino that cannot be reproduce on a computer.
But I'll try to find a solution, thanks for reporting it.

Your workaround is fine, additionally you can avoid a duplication with

```c++
root["result"] = statResponse.c_str();
```

beegee-tokyo, 2015-12-23:
First: I just downloaded your lib as zip. When I added it with 'Sketch'->Include Library->Add .ZIP Library...' there was no error message and I can use it without problems. It is listed properly in the Library Manager
```javascript
ArduinJson by Benoit Blanchon Version 5.0.7 INSTALLED
An efficient and elegant JSON library for Arduino. Like this project? Please star it on GitHub!
More info
```
Second: I am still learning about programming on low memory devices like the Arduino or ESP8266. I am spoiled by Android (and JAVA programming) with nearly no problems for program and variable memory. What should I use instead of String? char* ???

Third: Thanks for the tip with 
```javascript
root["result"] = statResponse.c_str();
```
I will give it a try.

bblanchon, 2015-12-23:
ESP8266 is not really a low memory device.
But small Arduino only have a couple of kilobytes of RAM, so you need to change your habit.

The secret is to avoid dynamic memory allocation on the heap and to only work on preallocated buffers on the stack, which is clearly an anti-pattern in Java.

You should have a look at this blog post of mine:
http://blog.benoitblanchon.fr/arduino-json-v5-0/

beegee-tokyo, 2015-12-23:
Off-topic (sorry)
Thanks for the link to your blog. I understand what you write there. But if I have a char*/String which has a size depending on an unknown response from a http client or server, how can I used a fixed size char*?

bblanchon, 2015-12-23:
You simply allocate a buffer for the worst case, and you make sure that a buffer overrun cannot happen.

If the response is too big for the buffer, it's up to you to cut it or reject it.

bblanchon, 2016-11-06:
Bug fixed in [ArduinoJson 5.7.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.7.0)

closed 2016-11-06 by bblanchon