bblanchon/ArduinoJson #187: Problem retrieve data after parsing

NitrofMtl, 2015-12-29:
Hi.

I have difficulty to retrieve data after parsing it, I don't know if it is only a misspelling but I can't fix it.

here the beginning of my data. I don't copy it entirely, it is a copy return by a pretttyPrintTo.
`"configs": [{
"K": 50,
"vs": 2.00,
"smm": 20.00,
"inputs": [{
....		`

so root a succes, if I do a print of root[configs], it print the content.

but I can't get anything of ;
`  K = root["configs"]["K"];`
  `  Serial.println(K);`
    `  vs = root["configs"]["vs"];`
    `  smm = root["configs"]["smm"];`

Thanks

bblanchon, 2015-12-29:
Hi,

`configs` is a nested array.

Solution:

```c++
K = root["configs"][0]["K"];
Serial.println(K);
vs = root["configs"][0]["vs"];
smm = root["configs"][0]["smm"];
```

You can improve code size and performance with:

```c++
JsonObject& config = root["configs"][0];
K = config["K"];
vs = config["vs"];
smm = config["smm"];
```

See [FAQ: Parsing succeeds but I can't read the values!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#parsing-succeeds-but-i-cant-read-the-values)

Regards

YooperKirks, 2015-12-29:
I believe you should have single quotes.   e.g.  
```python
root['config']['vs']
```

NitrofMtl, 2015-12-29:
Thanks. I knew that was that kind if thing.

btw you maybe should guide people in the wiki to use pretty print for object. I just catch the good way to use it ans it help me to understand better. also for the itherator command... and also the comande to retrieve 'it' count... I previously saw it in the issue... will find it back.

Thanks againt for the great work n Happy new year !

closed 2016-01-07 by bblanchon