bblanchon/ArduinoJson #188: Using ArduinoJson with gcc on Ubuntu

johannstieger, 2015-12-31:
Hi, I want to use my Arduino Client to transfer sensor data to a Ubuntu 15.10 or Synology Diskstation DS1813+ as server over a GSM shield with TCP. Transferring strings between the Arduino client and the server application works fine. When I want to use ArduinoJson to Transfer the data as JSON I can use it on my arduino without any problem.

But when I compilee the server-application with gcc and the ArduinoJson library included, I got a few error messages (file or folder not found). I think the problem is the ArduinoJson/include/arduinoJson/Arduino/String.hpp where there is an include of <WString.h> (and the WString.cpp) which are normally not on my Ubuntu machine. I have found these files on my Windows machine where I have installed the Arduino IDE and moved it onto my Ubuntu machine into the same folder as String.hpp and changed the <WString.h> to "WString.h" but I got many more error messages (maybe because other arduino libraries are missing - for example <avr/pgmspace.h>).

First of all I thought that I can use the same JSON library on my arduino and on my Linux server application (it's easier to understand because I'm a c/c++ newbie). On Arduino it works fine. Can you show me a way to use this library with gcc? I have tried to install gcc-avr and avr-libc but this didn't change anything.

Here you can see the error messages I got:
johann@johann-virtual-machine:~$ gcc server.c -L/usr/lib/mysql -lmysqlclient -I/usr/include/mysql -o server
In file included from ArduinoJson/include/ArduinoJson/Internals/../JsonBuffer.hpp:13:0,
                 from ArduinoJson/include/ArduinoJson/Internals/BlockJsonBuffer.hpp:9,
                 from ArduinoJson/include/ArduinoJson/DynamicJsonBuffer.hpp:9,
                 from ArduinoJson/include/ArduinoJson.h:7,
                 from ArduinoJson/ArduinoJson.h:13,
                 from server.c:40:
ArduinoJson/include/ArduinoJson/Internals/../Arduino/String.hpp:11:18: fatal error: string: File or folder not found
compilation terminated.

bblanchon, 2015-12-31:
Please try with `g++` instead of `gcc`.

bblanchon, 2015-12-31:
...I mean, use the C++ front-end of GCC, which is invoked by the `g++` command.

johannstieger, 2016-01-03:
Hi Benoit,
many thanks to this hint! Now I can compile my code with g++ despite when I add these two lines:
   #include "ArduinoJson/ArduinoJson.h"
   StaticJsonBuffer<200> jsonBuffer;
But when I add any other function - like these ones:
   JsonObject& json = jsonBuffer.createObject();
   json["Test"] = "This is a text-string";
I get error messages like these ones:
   undefined reference to: 'ArduinoJson::JsonBuffer::createObject()'
   undefined reference to: 'ArduinoJson::JsonObject::getOrCreateNodeAt(ArduinoJson::JsonObjectKey)'

What do I have to do to fix this error?

bblanchon, 2016-01-03:
Hi Johann,

Those are linker errors, you forgot to add something like `-lArduinoJson` on the g++ command line.
For this to work, you'll need another step to build the `ArduinoJson` library itself.

Since you're new to this world, I strongly advice that you use [CMake](https://cmake.org/) for this job.

All you need to write a `CMakeLists.txt` like this one:

```CMake
cmake_minimum_required(VERSION 2.8.12)
project(Server)

find_package(MySql)
add_subdirectory(ArduinoJson/src)

add_executable(server
    server.c
)

target_include_directories(server
    PRIVATE
    ${MYSQL_INCLUDE_DIR}
)

target_link_libraries(server
    ${MYSQL_LIBRARIES}
    ArduinoJson
)
```

Then just call:

```shell
mkdir build
cd build
cmake ..
make
```

This will create a `build/` folder then generate the `Makefile` for you and build.
I think very few people are still writing `Makefile`s by hand nowadays...

Good luck ;-)

closed 2016-01-13 by bblanchon