bblanchon/ArduinoJson #189: Unable to compile examples

trevstanhope, 2016-01-06:
I have installed the library as stated in your documentation to /usr/share/arduino running on a Debian 8.2 (Jessy), more specifically: Linux debian 3.16.0-4-686-pae #1 SMP Debian 3.16.7-ckt11-1+deb8u5 (2015-10-09) i686 GNU/Linux, g++-4.9, libstdc++6 and Arduino 1.0.5

However, I am unable to compile any of the example sketches. For example, I have received the following error when compiling JsonParserExample:

    JsonParserExample.cpp.o: In function `loop':
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to    `ArduinoJson::JsonBuffer::parseObject(char*, unsigned char)'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to    `ArduinoJson::JsonObject::getNodeAt(ArduinoJson::JsonObjectKey) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `char const*  ArduinoJson::JsonVariant::as<char const*>() const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonObject::getNodeAt(ArduinoJson::JsonObjectKey) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonVariant::asInteger() const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonObject::getNodeAt(ArduinoJson::JsonObjectKey) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::_invalid'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::_invalid'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::getNodeAt(unsigned int) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonVariant::asFloat() const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonObject::getNodeAt(ArduinoJson::JsonObjectKey) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::_invalid'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::_invalid'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonArray::getNodeAt(unsigned int) const'
    /usr/share/arduino/JsonParserExample.ino:38: undefined reference to `ArduinoJson::JsonVariant::asFloat() const'
    collect2: error: ld returned 1 exit status

... have you encountered this type of problem / do you have any suggestions you have to fix this error? I assume this is probably just a version support problem

bblanchon, 2016-01-06:
Looks like you forgot to link with the library.

Please have a look at #188.

trevstanhope, 2016-01-06:
I am encountering this error from the Arduino IDE itself, I am not compiling manually via g++ / gcc

bblanchon, 2016-01-06:
I'm sorry, I overlooked.

Arduino 1.0.5 requires [the "old-layout" package](https://github.com/bblanchon/ArduinoJson/releases/download/v5.0.7/ArduinoJson-v5.0.7-old-layout.zip), did you use this one ?

trevstanhope, 2016-01-06:
Problem solved!

I would suggest making a note of that in the docs, as Arduino 1.0.5 is still in the standard repos on Debian systems

bblanchon, 2016-01-06:
Done:
https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#arduion-10x-energia-stino-and-
https://github.com/bblanchon/ArduinoJson/wiki/Using-the-library-with-Arduino#for-older-arduino-ide

Feel free to edit the wiki.

closed 2016-01-06 by trevstanhope